coolsnowwolf/lede #313: 最新版本编译问题

amoboy, 2018-02-23:
更新的最新版本代码,一些设备比如720N,已经不再Generic里,看到issues里有说在Devices with small flash里.
编译后,bin里的文件

-rw-r--r-- 1 amo amo  14K 2月  23 14:36 config.seed
-rw-r--r-- 1 amo amo 7.0K 2月  23 14:43 openwrt-ar71xx-tiny-device-tl-wr720n-v3.manifest
-rw-r--r-- 1 amo amo 5.7M 2月  23 14:43 openwrt-ar71xx-tiny-root.squashfs
-rw-r--r-- 1 amo amo 1.4M 2月  23 14:43 openwrt-ar71xx-tiny-uImage-lzma.bin
-rwxr-xr-x 1 amo amo 4.3M 2月  23 14:43 openwrt-ar71xx-tiny-vmlinux.bin
-rwxr-xr-x 1 amo amo 4.3M 2月  23 14:43 openwrt-ar71xx-tiny-vmlinux.elf
-rw-r--r-- 1 amo amo 1.4M 2月  23 14:43 openwrt-ar71xx-tiny-vmlinux.lzma
-rwxr-xr-x 1 amo amo 1.4M 2月  23 14:43 openwrt-ar71xx-tiny-vmlinux-lzma.elf
-rw-r--r-- 1 amo amo  79M 2月  23 10:03 openwrt-imagebuilder-ar71xx-tiny.Linux-x86_64.tar.xz
-rw-r--r-- 1 amo amo  38M 2月  23 10:02 openwrt-sdk-ar71xx-tiny_gcc-5.5.0_musl.Linux-x86_64.tar.xz
-rw-r--r-- 1 amo amo  35M 2月  23 10:03 openwrt-toolchain-ar71xx-tiny_gcc-5.5.0_musl.Linux-x86_64.tar.bz2
drwxr-xr-x 2 amo amo  12K 2月  23 14:40 packages
-rw-r--r-- 1 amo amo 1.3K 2月  23 14:43 sha256sums
懵了,不知道要刷那个固件..
求解

amoboy, 2018-02-23:
编译没有错误,也没有出现too big...等..

amoboy, 2018-03-02:
修改完大小以后就可以编译了。

yuswtu, 2018-04-04:
怎么修改？去掉不需要的插件？

nimasi, 2018-04-15:
跟我一样啊 ，没有编译出正常的固件，怀疑是要改tiny-tp-link.mk的lzma压缩方式，等下试下

closed 2018-10-06 by coolsnowwolf