coolsnowwolf/lede #338: IPv6导致路由器自动重启

lijun0326, 2018-03-25:
K3，自行编译版本，使用默认设置+勾选了ebtables，尝试用ebtables搭网桥，使用如下命令：
```
ebtables -t broute -A BROUTING -p ! ipv6 -j DROP -i eth0.2
brctl addif br-lan eth0.2
```
然后v6正常，路由器几分钟以后会自动重启。

后面尝试直接使用把接入网线同时插到WAN和LAN口的方法，IPv6也正常，症状相同。
再后来使用全默认配置编译，使用上述方法依然不行。
同学的K2P+padavan在同一网络环境中使用上述命令工作正常。

lijun0326, 2018-03-25:
PS：我们学校使用的是静态IP，尝试上述方法前均已将WAN6删除，ULA前缀删除，odhcpd杀进程，LAN和WAN中路由通告等选项关闭。

LGA1150, 2018-04-22:
OpenWrt 下用 odhcpd relay 是最好的选择
/etc/config/dhcp
```
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option ra 'relay'
	option dhcpv6 'relay'
	option ndp 'relay'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config dhcp 'wan6'
	option interface 'wan'
	option ra 'relay'
	option dhcpv6 'relay'
	option ndp 'relay'
	option master '1'
```

lijun0326, 2018-04-22:
我用着odhcpd呢，他主要的问题在于，每过一会就用不了v6了，后台一直ping路由器可破（此处的“一会”在windows为几十分钟，*nix上为1-2分钟）

LGA1150, 2018-04-23:
你 odhcpd 版本多少？我 1.6

lijun0326, 2018-04-23:
1.3-1，感觉我需要更新（

lijun0326, 2018-04-23:
更新了一下，贼稳，不过为什么我用源码编译的，还会是旧版呢

LGA1150, 2018-04-23:
因为这个 repo 还没同步上游 openwrt/openwrt@fdf2e1fe1e952ff0c26da1e715d3f414f192dd9f

ZeaKyX, 2018-09-28:
新版本貌似不会了，另感谢你的代码

lijun0326, 2018-09-28:
@ZeaKyX 我这边最后测出来是K3的问题，换设备解决的（

ZeaKyX, 2018-09-29:
另外请教一下：
1、这个转发是不是与下面两个是同理的，很惭愧不太懂接口
https://www.jianshu.com/p/ed34cbf3b90c
https://i-meto.com/lede-ipv6/
2、为什么我开了之后ipv6加载非常缓慢，在padavan中不会如此？
3、能否实现开机启动呢，在padavan中我加在lan wan上线后开启，op不知道怎么搞

lijun0326, 2018-09-29:
@ZeaKyX 
1.跟jianshu那篇同理，属于无脑桥接。meto大佬那篇是用relay，我这里测试稳定性不行，各种崩
3. 我是写进rc.local的，大概如下
```
delayed_func()
{
    sleep 30s
    blabla
}
delayed_func&
```

closed 2018-10-06 by coolsnowwolf