coolsnowwolf/lede #343: Update version to R7.6以后，编译出错！e2fsprogs-1.43.7。求助！

d975xbx, 2018-03-30:
7.5以前都是顺利编译的。今天r7.6以后编译出错！求助！

Makefile:422: recipe for target 'create_inode.o' failed
make[6]: *** [create_inode.o] Error 1
make[6]: Leaving directory '/home/kevin/lede/build_dir/host/e2fsprogs-1.43.7/debugfs'
Makefile:404: recipe for target 'all-progs-recursive' failed
make[5]: *** [all-progs-recursive] Error 1
make[5]: Leaving directory '/home/kevin/lede/build_dir/host/e2fsprogs-1.43.7'
Makefile:339: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/kevin/lede/build_dir/host/e2fsprogs-1.43.7'
Makefile:50: recipe for target '/home/kevin/lede/build_dir/host/e2fsprogs-1.43.7/.built' failed
make[3]: *** [/home/kevin/lede/build_dir/host/e2fsprogs-1.43.7/.built] Error 2
make[3]: Leaving directory '/home/kevin/lede/tools/e2fsprogs'
tools/Makefile:150: recipe for target 'tools/e2fsprogs/compile' failed
make[2]: *** [tools/e2fsprogs/compile] Error 2
make[2]: Leaving directory '/home/kevin/lede'
tools/Makefile:148: recipe for target '/home/kevin/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyyyyynyyyyynyyynnyyyynyyyyyyyyyyyyyyyyyyynynnyyynny' failed
make[1]: *** [/home/kevin/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyyyyynyyyyynyyynnyyyynyyyyyyyyyyyyyyyyyyynynnyyynny] Error 2
make[1]: Leaving directory '/home/kevin/lede'
/home/kevin/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

d975xbx, 2018-03-31:
今天又好了，正常编译了。

feedeve, 2018-04-01:
我前几天都遇到了这个问题。

elvengod, 2018-04-03:
知道是因为什么原因么，我昨天也碰到这个问题了

closed 2018-10-06 by coolsnowwolf