coolsnowwolf/lede #348: lean大最新编译调整分区 overlay就不会加载

505575853, 2018-04-06:
lean大     最新编译发现只要调整
  x x    (64) Kernel partition size (in MB)                               x x
  x x    (192) Root filesystem partition size (in MB)                     x x
  这两个选项
  编译完后overlay就不加载了变成指向/tmp/root

.config - OpenWrt Configuration
 > Target Images qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
  lqqqqqqqqqqqqqqqqqqqqqqqqqqqqq Target Images qqqqqqqqqqqqqqqqqqqqqqqqqqqqqk
  x  Arrow keys navigate the menu.  <Enter> selects submenus ---> (or empty x
  x  submenus ----).  Highlighted letters are hotkeys.  Pressing <Y>        x
  x  includes, <N> excludes, <M> modularizes features.  Press <Esc><Esc> to x
  x  exit, <?> for Help, </> for Search.  Legend: [*] built-in  [ ]         x
  x lqqqq^(-)qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqk x
  x x    (0) Seconds to wait before booting the default entry             x x
  x x    [ ] Build VirtualBox image files (VDI)                           x x
  x x    [*] Build VMware image files (VMDK)                              x x
  x x    [ ] Build Hyper-V image files (VHD)                              x x
  x x    -*- Pad images to filesystem size (for JFFS2)                    x x
  x x    [ ] GZip images                                                  x x
  x x        *** Image Options ***                                        x x
  x x    (64) Kernel partition size (in MB)                               x x
  x x    (192) Root filesystem partition size (in MB)                     x x
  x x    ()  Root partition on target device                              x x
  x mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj x
  tqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqu
  x        <Select>    < Exit >    < Help >    < Save >    < Load >         x
  mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj

  
  
  
  Using username "root".
root@10.0.0.9's password:

BusyBox v1.27.2 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r363-912498c
 -----------------------------------------------------
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
tmpfs                  1002.2M      1.3M   1000.9M   0% /tmp
tmpfs                  1002.2M    740.0K   1001.5M   0% /tmp/root
overlayfs:/tmp/root    1002.2M    740.0K   1001.5M   0% /
tmpfs                   512.0K         0    512.0K   0% /dev
root@OpenWrt:~#

lm317379829, 2018-04-07:
VM下是吗？
我也出现了这种问题。

closed 2018-04-09 by 505575853

whlsxl, 2018-10-21:
问题依然存在啊

the55555day, 2018-12-10:
我也是同样的问题，而且ext4格式虽然可以正常使用，但得到的分区大小却不是指定的大小，如果指定128M，结果出来的有可能是480M，莫名其妙。

lifetyper, 2019-05-10:
原来如此，我就说怎么每次重启后我的配置就丢了，一查确实是overlay加载的问题，全部加载到ram上去了。