coolsnowwolf/lede #355: V2ray permission denied 无法运行

barton0426, 2018-04-11:
K3 R7.6.1版
安装后 在luci-app-v2ray中启动，一直显示未启动
在shell中运行v2ray 提示permission denied
chmod 777 v2ray后 依然permission denied
求助

lm317379829, 2018-04-13:
从github下载对应cpu的文件替换试试。

crackself, 2018-04-14:
shell 启动试试  ./usr/bin/v2ray/v2ray -config=/etc/v2ray/config.json 。
我通过命令可以运行，但是luci界面选项显示空白

lm317379829, 2018-04-14:
楼上应该是没有装luci-app-filetransfer。

crackself, 2018-04-14:
@lm317379829 luci-app-filetransfer不是写明“LuCI page for IPK upload”，用来上传ipk安装文件的吗，和v2ray运行有关?

clswang, 2018-04-14:
我的也不能运行，在/etc/v2ray/下面没有config.json是不是需要手动添加，luci-app-filetransfer查看已经安装。

crackself, 2018-04-14:
config.json是从config转换来的，要c-json软件包才可以，你看看有没有安装了

barton0426, 2018-04-14:
感谢crackself地解决方案 @crackself
能够运行了 不过总感觉哪里不太对 不如我自己修改的sspro速度快

clswang, 2018-04-15:
软件包里找不到lua-cjson, make menuconfig里也找不到，any advise.

barton0426, 2018-04-15:
update了吗？我刚刚安装的 就搜json试试

clswang, 2018-04-15:
opkg update后找到lua-cjson,thanks.

lm317379829, 2018-04-15:
@clswang  可以在.config文件中找找，或者直接在v2ray的makefile文件中添加对应依赖。

lm317379829, 2018-04-15:
@crackself 用来上传配置文件。

closed 2018-04-19 by barton0426

zq133, 2021-08-24:
把端口改一下就行了