coolsnowwolf/lede #361: 这个编译错误是什么问题？请指教！

shenzt68, 2018-04-17:
Makefile:344: recipe for target '/home/szt/openwrt-source/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/phicomm-k3-kernel.bin' failed
make[5]: *** [/home/szt/openwrt-source/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/phicomm-k3-kernel.bin] Error 1
make[5]: *** Waiting for unfinished jobs....
[======================|                                                                                                                                                                   ]  336/2727  12%mv /home/szt/openwrt-source/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin.new /home/szt/openwrt-source/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c5-v2-kernel.bin
mv /home/szt/openwrt-source/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c9-v1-kernel.bin.new /home/szt/openwrt-source/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/tplink-archer-c9-v1-kernel.bin
[=========================================================================================================================================================================================\] 2727/2727 100%
Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
        compressed data, compressed metadata, compressed fragments, no xattrs
        duplicates are removed
Filesystem size 18638.99 Kbytes (18.20 Mbytes)
        32.45% of uncompressed filesystem size (57435.77 Kbytes)
Inode table size 22610 bytes (22.08 Kbytes)
        21.95% of uncompressed inode table size (103024 bytes)
Directory table size 31074 bytes (30.35 Kbytes)
        45.29% of uncompressed directory table size (68614 bytes)
Number of duplicate files found 310
Number of inodes 3131
Number of files 2622
Number of fragments 110
Number of symbolic links  304
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 204
Number of ids (unique uids + gids) 1
Number of uids 1
        root (0)
Number of gids 1
        root (0)
make[5]: Leaving directory '/home/szt/openwrt-source/target/linux/bcm53xx/image'
Makefile:31: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/szt/openwrt-source/target/linux/bcm53xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/szt/openwrt-source/target/linux'
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/szt/openwrt-source'
target/Makefile:19: recipe for target '/home/szt/openwrt-source/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install' failed
make[1]: *** [/home/szt/openwrt-source/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/szt/openwrt-source'
Build failed - please re-run with -j1 to see the real error message
/home/szt/openwrt-source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 1

shenzt68, 2018-04-17:
自己学习给k3打补丁，然后编译的。只要软件包选多一些，就这样了。直接编译L大的源码不会出现些问题。

paldier, 2018-04-18:
目测超过了分区大小

closed 2018-10-06 by coolsnowwolf