coolsnowwolf/lede #366: 酸酸乳无法启动（无法保存）

qlwz, 2018-04-20:
路由器：Gen8 esxi6.0
问题：1、酸酸乳无法启动（无法保存，包括ip信息）
	  2、老版本的alidns 无法启动（昨天测试，本次重新编译并无加上）
	  
全新编译安装

git clone https://github.com/coolsnowwolf/lede
cd lede
./scripts/feeds update -a 
./scripts/feeds install -a
make menuconfig    
  只选择了  x86 x86_64  Generic

make -j1 V=s   

就这几个命令

编译完成后直接使用 StarWind Converter 将 openwrt-x86-64-combined-squashfs.img 转成 vmdk

`[    0.000000] Linux version 4.14.32 (qlwz@qlwz-virtual-machine) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r432-3415658)) #0 SMP Wed Apr 11 05:25:53 2018
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=fbbc8933-02 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009f7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003feeffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003fef0000-0x000000003fefefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003feff000-0x000000003fefffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003ff00000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffe0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] random: fast init done
[    0.000000] SMBIOS 2.4 present.
[    0.000000] DMI: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 09/21/2015
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 3292.520 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000000] vmware: using sched offset of 2424949025 ns
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x40000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-EFFFF uncachable
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 00000000000 mask 3FFC0000000 write-back
[    0.000000]   1 disabled
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] found SMP MP-table at [mem 0x000f6b30-0x000f6b3f] mapped at [ffffffffff400b30]
[    0.000000] Base memory trampoline at [ffff880000099000] 99000 size 24576
[    0.000000] BRK [0x01df7000, 0x01df7fff] PGTABLE
[... 1655 lines cut ...]
Fri Apr 20 23:49:07 2018 daemon.info dnsmasq[3636]: using local addresses only for domain invalid
Fri Apr 20 23:49:07 2018 daemon.info dnsmasq[3636]: using local addresses only for domain example.net
Fri Apr 20 23:49:07 2018 daemon.info dnsmasq[3636]: using local addresses only for domain example.org
Fri Apr 20 23:49:07 2018 daemon.info dnsmasq[3636]: using local addresses only for domain example.com
Fri Apr 20 23:49:07 2018 daemon.info dnsmasq[3636]: using nameserver 192.168.100.1#53
Fri Apr 20 23:49:07 2018 daemon.info dnsmasq[3636]: using 3 more local addresses
Fri Apr 20 23:49:07 2018 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Fri Apr 20 23:49:07 2018 user.notice firewall: Reloading firewall due to ifup of wan6 (eth1)
Fri Apr 20 23:49:07 2018 user.notice ddns-scripts[4432]: myddns_ipv6: PID '4432' started at 2018-04-20 23:49
Fri Apr 20 23:49:07 2018 user.notice firewall: Reloading firewall due to ifup of wan (eth1)
Fri Apr 20 23:49:07 2018 user.warn ddns-scripts[4432]: myddns_ipv6: Service section disabled! - TERMINATE
Fri Apr 20 23:49:07 2018 user.warn ddns-scripts[4432]: myddns_ipv6: PID '4432' exit WITH ERROR '1' at 2018-04-20 23:49
Fri Apr 20 23:49:07 2018 user.notice ddns-scripts[4668]: myddns_ipv4: PID '4668' started at 2018-04-20 23:49
Fri Apr 20 23:49:07 2018 user.warn ddns-scripts[4668]: myddns_ipv4: Service section disabled! - TERMINATE
Fri Apr 20 23:49:07 2018 user.warn ddns-scripts[4668]: myddns_ipv4: PID '4668' exit WITH ERROR '1' at 2018-04-20 23:49
Fri Apr 20 23:49:08 2018 daemon.notice netifd: bridge 'br-lan' link is up
Fri Apr 20 23:49:08 2018 daemon.notice netifd: Interface 'lan' has link connectivity
Fri Apr 20 23:49:08 2018 kern.info kernel: [   52.639985] br-lan: port 1(eth0) entered blocking state
Fri Apr 20 23:49:08 2018 kern.info kernel: [   52.640103] br-lan: port 1(eth0) entered forwarding state
Fri Apr 20 23:49:08 2018 kern.info kernel: [   52.640248] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
Fri Apr 20 23:49:08 2018 kern.info kernel: [   52.640568] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Fri Apr 20 23:49:10 2018 daemon.info dnsmasq[3636]: read /etc/hosts - 4 addresses
Fri Apr 20 23:49:10 2018 daemon.info dnsmasq[3636]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Fri Apr 20 23:49:18 2018 daemon.err uhttpd[2695]: luci: accepted login on / for root from 1.1.1.200
`

qlwz, 2018-04-20:
[    0.000000] Linux version 4.14.32 (qlwz@qlwz-virtual-machine) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r432-3415658)) #0 SMP Wed Apr 11 05:25:53 2018
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=fbbc8933-02 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009f7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003feeffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003fef0000-0x000000003fefefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003feff000-0x000000003fefffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003ff00000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffe0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] random: fast init done
[    0.000000] SMBIOS 2.4 present.
[    0.000000] DMI: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 09/21/2015
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 3292.520 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000000] vmware: using sched offset of 2424949025 ns
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x40000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-EFFFF uncachable
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 00000000000 mask 3FFC0000000 write-back
[    0.000000]   1 disabled
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] found SMP MP-table at [mem 0x000f6b30-0x000f6b3f] mapped at [ffffffffff400b30]
[    0.000000] Base memory trampoline at [ffff880000099000] 99000 size 24576
[    0.000000] BRK [0x01df7000, 0x01df7fff] PGTABLE
[    0.000000] BRK [0x01df8000, 0x01df8fff] PGTABLE
[    0.000000] BRK [0x01df9000, 0x01df9fff] PGTABLE
[    0.000000] BRK [0x01dfa000, 0x01dfafff] PGTABLE
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F6AC0 000024 (v02 PTLTD )
[    0.000000] ACPI: XSDT 0x000000003FEF114C 00005C (v01 INTEL  440BX    06040000 VMW  01324272)
[    0.000000] ACPI: FACP 0x000000003FEFEE73 0000F4 (v04 INTEL  440BX    06040000 PTL  000F4240)
[    0.000000] ACPI: DSDT 0x000000003FEF13B4 00DABF (v01 PTLTD  Custom   06040000 MSFT 03000001)
[    0.000000] ACPI: FACS 0x000000003FEFFFC0 000040
[    0.000000] ACPI: FACS 0x000000003FEFFFC0 000040
[    0.000000] ACPI: BOOT 0x000000003FEF138C 000028 (v01 PTLTD  $SBFTBL$ 06040000  LTP 00000001)
[    0.000000] ACPI: APIC 0x000000003FEF133C 000050 (v01 PTLTD  ? APIC   06040000  LTP 00000000)
[    0.000000] ACPI: MCFG 0x000000003FEF1300 00003C (v01 PTLTD  $PCITBL$ 06040000  LTP 00000001)
[    0.000000] ACPI: SRAT 0x000000003FEF1248 0000B8 (v02 VMWARE MEMPLUG  06040000 VMW  00000001)
[    0.000000] ACPI: HPET 0x000000003FEF1210 000038 (v01 VMWARE VMW HPET 06040000 VMW  00000001)
[    0.000000] ACPI: WAET 0x000000003FEF11E8 000028 (v01 VMWARE VMW WAET 06040000 VMW  00000001)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[... 968 lines cut ...]
[   22.898790] br-lan: port 1(eth0) entered forwarding state
[   22.898920] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   22.899932] vmxnet3 0000:13:00.0 eth1: intr type 3, mode 0, 2 vectors allocated
[   22.900463] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
[   22.900895] 8021q: adding VLAN 0 to HW filter on device eth1
[   25.217190] random: crng init done
[   50.522678] br-lan: port 1(eth0) entered disabled state
[   50.523584] device eth0 left promiscuous mode
[   50.523901] br-lan: port 1(eth0) entered disabled state
[   50.525037] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   51.662509] vmxnet3 0000:0b:00.0 eth0: intr type 3, mode 0, 2 vectors allocated
[   51.663072] vmxnet3 0000:0b:00.0 eth0: NIC Link is Up 10000 Mbps
[... cut ...]
[   51.665728] vmxnet3 0000:13:00.0 eth1: intr type 3, mode 0, 2 vectors allocated
[   51.666273] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
[   51.666566] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   51.666680] 8021q: adding VLAN 0 to HW filter on device eth1
[   52.639985] br-lan: port 1(eth0) entered blocking state
[   52.640103] br-lan: port 1(eth0) entered forwarding state
[   52.640248] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   52.640568] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

qlwz, 2018-04-20:
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.162190] pci 0000:00:0f.0: BAR 6: assigned [mem 0x40800000-0x40807fff pref]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.162330] pci 0000:00:15.3: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.162448] pci 0000:00:15.3: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.162570] pci 0000:00:15.4: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.162687] pci 0000:00:15.4: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.162810] pci 0000:00:15.5: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.162927] pci 0000:00:15.5: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.163050] pci 0000:00:15.6: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.163166] pci 0000:00:15.6: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.163289] pci 0000:00:15.7: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.163406] pci 0000:00:15.7: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.163534] pci 0000:00:16.3: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.163651] pci 0000:00:16.3: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.163773] pci 0000:00:16.4: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.163890] pci 0000:00:16.4: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.164013] pci 0000:00:16.5: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.164130] pci 0000:00:16.5: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.164253] pci 0000:00:16.6: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.164369] pci 0000:00:16.6: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.164492] pci 0000:00:16.7: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.164609] pci 0000:00:16.7: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.164732] pci 0000:00:17.3: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.164849] pci 0000:00:17.3: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.164971] pci 0000:00:17.4: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.165088] pci 0000:00:17.4: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.165211] pci 0000:00:17.5: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.165328] pci 0000:00:17.5: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.165451] pci 0000:00:17.6: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.165567] pci 0000:00:17.6: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.165693] pci 0000:00:17.7: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.165810] pci 0000:00:17.7: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.165933] pci 0000:00:18.2: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.166050] pci 0000:00:18.2: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.166173] pci 0000:00:18.3: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.166290] pci 0000:00:18.3: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.166413] pci 0000:00:18.4: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.166530] pci 0000:00:18.4: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.166652] pci 0000:00:18.5: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.166769] pci 0000:00:18.5: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.166892] pci 0000:00:18.6: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.167009] pci 0000:00:18.6: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.167141] pci 0000:00:18.7: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.167259] pci 0000:00:18.7: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.167386] pci 0000:00:18.7: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.167503] pci 0000:00:18.7: BAR 7: failed to assign [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.167627] pci 0000:00:18.6: BAR 7: no space for [io  size 0x1000]
Fri Apr 20 23:48:38 2018 kern.info kernel: [    0.167744] pci 0000:00:18.6: BAR 7: failed to assign [io  size 0x1000]
[... cut ...]
Fri Apr 20 23:49:08 2018 kern.info kernel: [   52.640248] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
Fri Apr 20 23:49:08 2018 kern.info kernel: [   52.640568] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Fri Apr 20 23:49:10 2018 daemon.info dnsmasq[3636]: read /etc/hosts - 4 addresses
Fri Apr 20 23:49:10 2018 daemon.info dnsmasq[3636]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Fri Apr 20 23:49:18 2018 daemon.err uhttpd[2695]: luci: accepted login on / for root from 1.1.1.200

a843161264, 2018-04-30:
好像是luci的问题，不只是ssr，其他配置同样保存&应用无效，需要额外点击一次应用。另，ssr的问题配置好以后应用&重启即可启动。

closed 2018-10-06 by coolsnowwolf