coolsnowwolf/lede #375: package/lean/luci-app-syncdial 缺乏依赖

tossp, 2018-04-23:
切换到`src-git luci https://git.lede-project.org/project/luci.git;lede-17.01`之后

`WARNING: Makefile 'package/lean/luci-app-syncdial/Makefile' has a dependency on 'luci-app-mwan3', which does not exist`

gavin219, 2018-04-23:
lede-17.01的mwan3名字不一样，可以删除tmp后重新update一下

tossp, 2018-04-23:
全新配置环境也是失败的
@gavin219

tossp, 2018-04-23:
```
 make[3] -C /cache/zh-lede-k3-ci/feeds/luci/modules/luci-mod-admin-full compile
 make[3] -C /cache/zh-lede-k3-ci/feeds/packages/net/sqm-scripts compile
 make[3] -C /cache/zh-lede-k3-ci/feeds/luci/collections/luci compile
 make[3] -C /cache/zh-lede-k3-ci/feeds/luci/applications/luci-app-ddns compile
 make[3] -C package/lean/luci-lib-fs compile
 make[3] -C package/lean/luci-app-autoreboot compile
 make[3] -C package/lean/luci-app-filetransfer compile
 make[2] package/install
make[2]: warning: -jN forced in submake: disabling jobserver mode.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-syncdial:
 * 	luci-app-mwan3 * 
 * opkg_install_cmd: Cannot install package luci-app-syncdial.
make[2]: *** [package/install] Error 255
make[1]: *** [/builds/eeb30c0e/1/zh/lede-k3/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_install] Error 2
make -r world: build failed. Please re-run make with -j1 V=s to see what's going on
make: *** [world] Error 1
```

closed 2018-04-26 by tossp