coolsnowwolf/lede #387: 编译R7.7出错

schen39, 2018-04-26:
已经重复clone源码N次了，酸酸乳也开了全局，还是tar错误，信息如下：

renameat2.c:81:22: note: each undeclared identifier is reported only once for each function it appears in
make[8]: *** [renameat2.o] Error 1
make[8]: *** Waiting for unfinished jobs....
make[8]: Leaving directory `/home/schen39/lede/build_dir/host/tar-1.30/gnu'
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory `/home/schen39/lede/build_dir/host/tar-1.30/gnu'
make[6]: *** [all] Error 2
make[6]: Leaving directory `/home/schen39/lede/build_dir/host/tar-1.30/gnu'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/schen39/lede/build_dir/host/tar-1.30'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/schen39/lede/build_dir/host/tar-1.30'
make[3]: *** [/home/schen39/lede/build_dir/host/tar-1.30/.built] Error 2
make[3]: Leaving directory `/home/schen39/lede/tools/tar'
make[2]: *** [tools/tar/compile] Error 2
make[2]: Leaving directory `/home/schen39/lede'
make[1]: *** [/home/schen39/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnyyyynyyyyyyyyyyyyyyyyyynynnyyynny] Error 2
make[1]: Leaving directory `/home/schen39/lede'
Build failed - please re-run with -j1 to see the real error message
make: *** [world] 错误 1

o213o0, 2018-08-02:
可能你的ubuntu版本是**Ubuntu 14.04.1 LTS**，需要### apt-get upgrade, 到	**Ubuntu 14.04.5 LTS**有些问题就解决了。

mailsex, 2018-08-11:
collect2: error: ld returned 1 exit status
make[6]: *** [cc1plus] Error 1
make[6]: Leaving directory `/root/ledeild_dir/toolchain-arm_cortex-a9_gcc-7.3.0_musl_eabi/gcc-7.3.0-final/gcc'
make[5]: *** [all-gcc] Error 2
make[5]: Leaving directory `/root/ledeild_dir/toolchain-arm_cortex-a9_gcc-7.3.0_musl_eabi/gcc-7.3.0-final'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/root/ledeild_dir/toolchain-arm_cortex-a9_gcc-7.3.0_musl_eabi/gcc-7.3.0-final'
make[3]: *** [/root/ledeild_dir/toolchain-arm_cortex-a9_gcc-7.3.0_musl_eabi/gcc-7.3.0-final/.built] Error 2
make[3]: Leaving directory `/root/lede/toolchain/gcc/final'
make[2]: *** [toolchain/gcc/final/compile] Error 2
make[2]: Leaving directory `/root/lede'
make[1]: *** [/root/lede/staging_dir/toolchain-arm_cortex-a9_gcc-7.3.0_musl_eabi/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory `/root/lede'
make: *** [world] Error 2

同样错误。。。

akari263, 2018-08-12:
> renameat2.c: In function 'renameat2':
renameat2.c:81:22: error: '__NR_renameat2' undeclared (first use in this function)
   ret_val = syscall (SYS_renameat2, fd1, src, fd2, dst, flags);
                      ^
renameat2.c:81:22: note: each undeclared identifier is reported only once for each function it appears in
make[7]: *** [renameat2.o] Error 1
make[7]: Leaving directory `/home/ubuntu64/lede/build_dir/host/patch-2.7.6/lib'
make[6]: *** [all] Error 2
make[6]: Leaving directory `/home/ubuntu64/lede/build_dir/host/patch-2.7.6/lib'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/ubuntu64/lede/build_dir/host/patch-2.7.6'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/ubuntu64/lede/build_dir/host/patch-2.7.6'
make[3]: *** [/home/ubuntu64/lede/build_dir/host/patch-2.7.6/.built] Error 2
make[3]: Leaving directory `/home/ubuntu64/lede/tools/patch'
make[2]: *** [tools/patch/compile] Error 2
make[2]: Leaving directory `/home/ubuntu64/lede'
make[1]: *** [/home/ubuntu64/lede/staging_dir/target-mips_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnyyyyyyyyyyyyyyyyyyyynyynynyyyynny] Error 2
make[1]: Leaving directory `/home/ubuntu64/lede'
make: *** [world] 错误 2

也是这个错误

closed 2018-10-06 by coolsnowwolf