coolsnowwolf/lede #417: 今天git的源码，编译不过。以前都行的。

shenzt68, 2018-05-24:
| recommend you do so.                                        |
|                                                             |
| GMP has been carefully tested by its authors, but compilers |
| are all too often released with serious bugs.  GMP tends to |
| explore interesting corners in compilers and has hit bugs   |
| on quite a few occasions.                                   |
|                                                             |
+-------------------------------------------------------------+

make[8]: Leaving directory '/home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2'
make[7]: Leaving directory '/home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2'
make[6]: Leaving directory '/home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2'
make[5]: Leaving directory '/home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2'
make[4]: Leaving directory '/home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2'
touch /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/.built
rm -rf /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/.pkgdir/libgmp.installed /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/.pkgdir/libgmp
mkdir -p /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/.pkgdir/libgmp
install -d -m0755 /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/.pkgdir/libgmp/usr/lib
cp -fpR /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-install/usr/lib/libgmp.so.* /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/.pkgdir/libgmp/usr/lib/
touch /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/.pkgdir/libgmp.installed
mkdir -p /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp
SHELL= flock /home/szt/lede/tmp/.root-copy.flock -c 'cp -fpR /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/.pkgdir/libgmp/. /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.libgmp_installed
mkdir -p /home/szt/lede/bin/targets/bcm53xx/generic/packages /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-arm_cortex-a9/libgmp/CONTROL /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo
install -d -m0755 /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-arm_cortex-a9/libgmp/usr/lib
cp -fpR /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-install/usr/lib/libgmp.so.* /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-arm_cortex-a9/libgmp/usr/lib/
find /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-arm_cortex-a9/libgmp -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="arm-openwrt-linux-muslgnueabi-"   ; NM="arm-openwrt-linux-muslgnueabi-nm" STRIP="/home/szt/lede/staging_dir/host/bin/sstrip" STRIP_KMOD="/home/szt/lede/scripts/strip-kmod.sh" PATCHELF="/home/szt/lede/staging_dir/host/bin/patchelf" /home/szt/lede/scripts/rstrip.sh /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-arm_cortex-a9/libgmp
rstrip.sh: /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-arm_cortex-a9/libgmp/usr/lib/libgmp.so.10.3.2: shared object
(cd /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-arm_cortex-a9/libgmp/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/szt/lede/bin/packages/arm_cortex-a9/base
/home/szt/lede/scripts/ipkg-build -c -o 0 -g 0 /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-arm_cortex-a9/libgmp /home/szt/lede/bin/packages/arm_cortex-a9/base
Packaged contents of /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-arm_cortex-a9/libgmp into /home/szt/lede/bin/packages/arm_cortex-a9/base/libgmp_6.1.2-1_arm_cortex-a9.ipk
echo "libgmp" >> /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/gmp.default.install
rm -rf /home/szt/lede/tmp/stage-gmp
mkdir -p /home/szt/lede/tmp/stage-gmp/host /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/packages /home/szt/lede/staging_dir/host/packages
install -d -m0755 /home/szt/lede/tmp/stage-gmp/usr/include
cp -fpR /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-install/usr/include/gmp* /home/szt/lede/tmp/stage-gmp/usr/include/
install -d -m0755 /home/szt/lede/tmp/stage-gmp/usr/lib
cp -fpR /home/szt/lede/build_dir/target-arm_cortex-a9_musl_eabi/gmp-6.1.2/ipkg-install/usr/lib/libgmp.{a,so*} /home/szt/lede/tmp/stage-gmp/usr/lib/
find /home/szt/lede/tmp/stage-gmp -name '*.la' | xargs -r rm -f;        find /home/szt/lede/tmp/stage-gmp -name '*.la' | xargs -r rm -f;
if [ -f /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/packages/gmp.list ]; then /home/szt/lede/scripts/clean-package.sh "/home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/packages/gmp.list" "/home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi"; fi
if [ -d /home/szt/lede/tmp/stage-gmp ]; then (cd /home/szt/lede/tmp/stage-gmp; find ./ > /home/szt/lede/tmp/stage-gmp.files);   SHELL= flock /home/szt/lede/tmp/.staging-dir.flock -c ' mv /home/szt/lede/tmp/stage-gmp.files /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/packages/gmp.list && cp -fpR /home/szt/lede/tmp/stage-gmp/* /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/; '; fi
rm -rf /home/szt/lede/tmp/stage-gmp
touch /home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.gmp_installed
make[3]: Leaving directory '/home/szt/lede/package/libs/gmp'
make[2]: Leaving directory '/home/szt/lede'
package/Makefile:103: recipe for target '/home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/szt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/szt/lede'
Build failed - please re-run with -j1 to see the real error message
/home/szt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 1

shenzt68, 2018-05-26:
重新git ，仍然不过

kill33, 2018-06-01:
Build failed - please re-run with -j1 to see the real error message

用 make -j1 V=s  编一下看看真是报错信息

closed 2018-10-06 by coolsnowwolf