coolsnowwolf/lede #418: qemu编译报错

wheretoteardown, 2018-05-24:
CHK version_gen.h
make[5]: 'libfdt/libfdt.a' is up to date.
make[5]: Leaving directory '/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/dtc'
gcc -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/util -Iutil -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/tcg -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/tcg/i386 -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/linux-headers -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/linux-headers -I. -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1 -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/accel/tcg -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/include -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/dtc/libfdt -pthread -I/home/yhk/lede/staging_dir/hostpkg/include/glib-2.0 -I/home/yhk/lede/staging_dir/hostpkg/lib/glib-2.0/include -DNCURSES_WIDECHAR  -m64 -mcx16 -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -Wstrict-prototypes -Wredundant-decls -Wall -Wundef -Wwrite-strings -Wmissing-prototypes -fno-strict-aliasing -fno-common -fwrapv  -Wexpansion-to-defined -Wendif-labels -Wno-shift-negative-value -Wno-missing-include-dirs -Wempty-body -Wnested-externs -Wformat-security -Wformat-y2k -Winit-self -Wignored-qualifiers -Wold-style-declaration -Wold-style-definition -Wtype-limits -I/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/tests -MMD -MP -MT util/memfd.o -MF util/memfd.d -O2 -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -O2 -I/home/yhk/lede/staging_dir/host/include -I/home/yhk/lede/staging_dir/hostpkg/include -I/home/yhk/lede/staging_dir/target-x86_64_musl/host/include  -c -o util/memfd.o util/memfd.c
util/memfd.c:40:12: error: static declaration of 'memfd_create' follows non-static declaration
 static int memfd_create(const char *name, unsigned int flags)
            ^~~~~~~~~~~~
In file included from /usr/include/bits/mman-linux.h:115,
                 from /usr/include/bits/mman.h:45,
                 from /usr/include/sys/mman.h:41,
                 from /home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/include/sysemu/os-posix.h:29,
                 from /home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/include/qemu/osdep.h:104,
                 from util/memfd.c:28:
/usr/include/bits/mman-shared.h:46:5: note: previous declaration of 'memfd_create' was here
 int memfd_create (const char *__name, unsigned int __flags) __THROW;
     ^~~~~~~~~~~~
make[4]: *** [/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/rules.mak:66: util/memfd.o] Error 1
make[4]: Leaving directory '/home/yhk/lede/build_dir/hostpkg/qemu-2.11.1'
make[3]: *** [Makefile:141: /home/yhk/lede/build_dir/hostpkg/qemu-2.11.1/.built] Error 2
make[3]: Leaving directory '/home/yhk/lede/package/lean/qemu-userspace'
make[2]: *** [package/Makefile:107: package/lean/qemu-userspace/host/compile] Error 2
make[2]: Leaving directory '/home/yhk/lede'
make[1]: *** [package/Makefile:103: /home/yhk/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yhk/lede'
make: *** [/home/yhk/lede/include/toplevel.mk:218: world] Error 2

legendtang, 2018-06-06:
遇到了相同的问题，编译 K2P 指定了一些 luci 包

legendtang, 2018-06-06:
http://lists.openembedded.org/pipermail/openembedded-core/2018-January/146949.html
上游有修复，不知道是否合入 18.06 中，似乎是 glibc 高版本的问题

closed 2018-10-06 by coolsnowwolf