coolsnowwolf/lede #432: 最新代码编译出错

hoseahsu, 2018-06-05:
下面是最后面的错误输出，麻烦看一下是什么问题
```
PATH=/home/hosea/github/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/bin:/home/hosea/github/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/bin:/home/hosea/github/lede/staging_dir/host/bin:/home/hosea/github/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/bin:/home/hosea/github/lede/staging_dir/host/bin:/home/hosea/github/lede/staging_dir/host/bin:/home/hosea/anaconda3/bin:/home/hosea/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
PKG_CONFIG_PATH=/home/hosea/github/lede/staging_dir/host/lib/pkgconfig:/home/hosea/github/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/lib/pkgconfig
SCAN_COOKIE=11280
MAKEFLAGS=
SESSION_MANAGER=local/XuZL:@/tmp/.ICE-unix/1854,unix/XuZL:/tmp/.ICE-unix/1854
HOST_OS=Linux
BUILD_VARIANT=
GTK_IM_MODULE=fcitx
yes '' | make -C /home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-4.9.105 HOSTCFLAGS="-O2 -I/home/hosea/github/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="" ARCH="um" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Jun  4 08:01:52 2018" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/hosea/github/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.9.105 CC="gcc" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result" oldconfig
make[4]: Entering directory '/home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-4.9.105'
scripts/kconfig/conf  --oldconfig arch/x86/um/Kconfig
net/sched/Kconfig:44: warning: menuconfig statement without prompt
kernel/time/Kconfig:155:warning: range is invalid
#
# configuration written to .config
#
make[4]: Leaving directory '/home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-4.9.105'
mkdir -p /home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-dev
make -C /home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-4.9.105 HOSTCFLAGS="-O2 -I/home/hosea/github/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="" ARCH="um" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Jun  4 08:01:52 2018" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/hosea/github/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.9.105 CC="gcc" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result" INSTALL_HDR_PATH="/home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-dev/" headers_install
make[4]: Entering directory '/home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-4.9.105'
  CHK     include/generated/uapi/linux/version.h
Makefile:1165: *** Headers not exportable for the um architecture.  Stop.
make[4]: Leaving directory '/home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-4.9.105'
Makefile:104: recipe for target '/home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-4.9.105/.configured' failed
make[3]: *** [/home/hosea/github/lede/build_dir/toolchain-x86_64_gcc-7.3.0_musl/linux-4.9.105/.configured] Error 2
make[3]: Leaving directory '/home/hosea/github/lede/toolchain/kernel-headers'
toolchain/Makefile:96: recipe for target 'toolchain/kernel-headers/compile' failed
make[2]: *** [toolchain/kernel-headers/compile] Error 2
make[2]: Leaving directory '/home/hosea/github/lede'
toolchain/Makefile:94: recipe for target '/home/hosea/github/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/hosea/github/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/hosea/github/lede'
/home/hosea/github/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

hoseahsu, 2018-06-05:
就在`make menuconfig`时改了`target System`为 `user Mode Linux`，是想编一个64位的

mrclock, 2018-06-07:
64位的
Target System 选择X86
Subtarget选择x86_64

这样编译的就是64位版本了。Generic是32位的。

closed 2018-10-06 by coolsnowwolf