coolsnowwolf/lede #442: 重新下载源码编译发现menuconfig里面sfe不显示了，以前编译的还有

nimasi, 2018-06-15:
今天打算重新下载源码编译，结果更新了feeds后进menuconfig里面，sfe不显示了，文件夹里面shortcut-fe和luci-app-sfe都还在也没有缺少，重新下载编译还是没出现，请教下有解决办法吗

kill33, 2018-06-15:
源码目录运行 rm tmp -rf 试试

nimasi, 2018-06-15:
@kill33
删了tmp 还是一样的

LGA1150, 2018-06-15:
4.14 内核不需要 SFE 了

formjcdq, 2018-06-15:
@LGA1150 
flow offload看起来会使用ip6tables实现的IPv6 NAT失效，SFE不会。而IPv6 NAT在教育网中用的比较多（odhcpd总是有问题……），取消sfe，这点也挺麻烦的。

formjcdq, 2018-06-15:
@nimasi 
SFE被有意去除了，可以看SFE模块相关的Makefile，在4.14内核下就是不会出现。我尝试过修改Makefile，让它出现，但编译起来会报错，估计也确实有问题还没解决。鉴于4.14又有flow offload了，SFE也许永远不会兼容4.14了……

LGA1150, 2018-06-15:
@formjcdq
“看起来”。你测试过了？实际上并不会
odhcpd 问题请用最新源码编译 https://git.openwrt.org/project/odhcpd.git 修改 `package/network/services/odhcpd/Makefile` 把 `PKG_SOURCE_VERSION` 改为最新的 commit id

coolsnowwolf, 2018-06-15:
实际上 SFE 在 4.14 内核已经搞好了，但不打算继续用了，还是跟着官方走吧，维护起来比较方便

coolsnowwolf, 2018-06-15:
@formjcdq 实际上我还是保留了 4.9 内核的支持的。你不如修改 Makefile 切换回到 4.9 内核

formjcdq, 2018-06-15:
@LGA1150 之所以说是看起来，无非是因为，这只是我个人的测试结果，开了就用不了，关了就能用，不必想太多。我是担心也许是我个人问题导致的，也许你可以指点我下该怎么使offload和IPv6 NAT共存。odhcpd不想用只是因为之前试了太多版本了，都不如NAT稳定，所以我比较希望用NAT

LGA1150, 2018-06-15:
@formjcdq 
路由器型号？我在 ar71xx 的 WNDR3800 上测试正常
```
root@OpenWrt:~# grep -i ipv6.*OFFLOAD /proc/net/nf_conntrack
ipv6     10 tcp      6 src=2001:0470:000d:*:6bc1:4425:6056:91a3 dst=2404:6800:4008:0801:0000:0000:0000:200a sport=38692 dport=443 packets=3 bytes=741 src=2404:6800:4008:0801:0000:0000:0000:200a dst=2402:f000:0001:1501:0000:5efe:*:* sport=443 dport=38692 packets=2 bytes=152 [OFFLOAD] mark=0 zone=0 use=3
ipv6     10 udp      17 src=2001:0470:000d:*:6bc1:4425:6056:91a3 dst=2404:6800:4008:0802:0000:0000:0000:200e sport=51419 dport=443 packets=2 bytes=2563 src=2404:6800:4008:0802:0000:0000:0000:200e dst=2402:f000:0001:1501:0000:5efe:*:* sport=443 dport=51419 packets=1 bytes=1398 [OFFLOAD] mark=0 zone=0 use=3
```

注意源地址从 `2001:470::` 变成了 `2402:f000::` 证明 NAT 生效，OFFLOAD 标志也说明已被 OFFLOAD 加速

formjcdq, 2018-06-15:
@LGA1150 你说的对。我后来又测试了一次，开启offload不影响V6 NAT。可能是我第一次测试有别的原因。只是v4可以查到offload的记录，v6没有查到……设备是k2t，大雕的这个固件最新。

LGA1150, 2018-06-15:
@formjcdq 
如果你仔细观察这个插件的 init.d，会发现它根本没有调用 ip6tables

LGA1150, 2018-06-15:
@coolsnowwolf 
luci 从 https://github.com/openwrt/luci/commit/aadc1f79c1c5d782ed243955cbe2c8a60eb2a2ca 开始已经支持在 网络-防火墙 设置 offload，这个插件没有存在的必要性了

LGA1150, 2018-06-29:
@ItsLucas 
路由器上的 SSR？SFE/flowoffload 只能加速经路由器转发的数据包，用了透明代理后相当于路由器本身发起了数据包，是无法享受加速的

closed 2018-10-06 by coolsnowwolf

windyin, 2019-01-03:
X86 编译 menuconfig 里也不显示 luci-app-sfe了，奇怪

sqliuchang, 2019-06-30:
原来如此 怪不得切换内核后SFE没了