coolsnowwolf/lede #453: UPnP 不能使用

zhangchioulin, 2018-06-23:
## 问题
* 表现为Xbox one 显示“UPnP not successful in your network settings”
* shell 中`upnpc -s`输出

MxCen, 2018-06-25:
我这和你差不多情况，哇你写的真详细。

tossp, 2018-06-25:
嗯，应该是mwan作怪，整体表现为，内网正常，但无法连接外网
猜测和今天`208.67.222.222`突然被GFW阻断有关

看日志，猜测应该是mwan，判定wan口掉线后，阻断了wan口的使用

你可以尝试`/cgi-bin/luci/admin/network/mwan`去修改wan口的监测配置

最后附上我调整过的部分`/etc/config/mwan3`做参考，其中`61.139.2.69`和`218.6.200.139`是四川电信的dns服务器

```
......
config interface 'wan'
	option enabled '1'
	option family 'ipv4'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'
	option initial_state 'online'
	option track_method 'ping'
	option size '56'
	option check_quality '0'
	option failure_interval '5'
	option recovery_interval '5'
	list track_ip '61.139.2.69'
	list track_ip '218.6.200.139'
	list track_ip '114.114.114.114'
	list track_ip '114.114.115.115'
	list track_ip '223.5.5.5'
	list track_ip '223.6.6.6'
	list track_ip '119.29.29.29'
	list track_ip '1.2.4.8'
	list track_ip '8.8.8.8'
	option reliability '1'
	option flush_conntrack 'never'
......
```

closed 2018-10-06 by coolsnowwolf

WordlessEcho, 2020-01-10:
运行如下指令：
```shell
touch /var/run/miniupnpd.leases
```