coolsnowwolf/lede #458: v2ray启动错误

lijun0326, 2018-06-26:
```
root@OpenWrt:/etc/init.d# ./v2raypro start
/usr/bin/lua: /etc/v2ray/gen_config.lua:12: module 'cjson.safe' not found:
        no field package.preload['cjson.safe']
        no file './cjson/safe.lua'
        no file '/usr/share/lua/cjson/safe.lua'
        no file '/usr/share/lua/cjson/safe/init.lua'
        no file '/usr/lib/lua/cjson/safe.lua'
        no file '/usr/lib/lua/cjson/safe/init.lua'
        no file './cjson/safe.so'
        no file '/usr/lib/lua/cjson/safe.so'
        no file '/usr/lib/lua/loadall.so'
        no file './cjson.so'
        no file '/usr/lib/lua/cjson.so'
        no file '/usr/lib/lua/loadall.so'
stack traceback:
        [C]: in function 'require'
        /etc/v2ray/gen_config.lua:12: in main chunk
        [C]: ?
V2Ray v3.26 (die Commanderin) 20180614
A unified platform for anti-censorship.
Main: failed to load config: /etc/v2ray/config.json > Main|ConfLoader|External: config file not readable > open /etc/v2ray/config.json: no such file or directory
```

lijun0326, 2018-06-26:
看起来是选中v2ray的时候没有强制依赖`lua-cjson`的问题

jzhou404, 2018-07-16:
我安装了lua-cjson_2.1.0-2_i386_pentium4.ipk，V2ray就可以启动正常了，就是编译的时候lua-cjson默认是M的， 而不是强制编译进去的原因

closed 2018-10-06 by coolsnowwolf