coolsnowwolf/lede #465: ssr可否添加保存多个服务器功能？

nimasi, 2018-06-28:


lyin888, 2018-07-09:
HAProxy
HAProxy能够检测Shadowsocks服务器的连通情况，从而实现负载均衡和高可用的功能，支持主备用服务器宕机自动切换，并且可以设置多个主服务器用于分流，规定每个分流节点的流量比例等。前提条件是你的所有Shadowsocks服务器的【加密方式】和【密码】一致。

使用方法：配置好你的Shadowsocks服务器ip地址和端口，然后开启Shadowsocks服务，将服务器地址填写为【127.0.0.1】，端口【2222】，其他参数和之前一样即可，你可以通过访问【路由器的IP:1111/haproxy】输入用户名admin，密码root来观察各节点健康状况，红色为宕机，绿色正常,使用说明请点击这里

后台监控页面：192.168.1.1 :1111/haproxy 用户名admin，密码root

lyin888, 2018-07-09:
[right.com.cn/forum/thread-198649-1-1.html](url)

nimasi, 2018-07-09:
@lyin888
https://github.com/AlexZhuo/luci-app-haproxy-tcp
是这个吗 图片上面只看到ip 端口设置 能设置密码 协议 之类的吗

lyin888, 2018-07-09:
LEAN固件包含HAProxy，编译的时候选择上就可以

lyin888, 2018-07-10:
lean固件在选项Utilities/Luci下，勾选编译就可以了

WilllllChen, 2018-07-10:
@lyin888 
我也有同样的需求，有时候只是因为游戏或者其他原因想切换一下服务器，并不想做负载均衡，况且HA要求加密方式一样不太现实。每换一次就要重新手输配置太不方便了..............当然这要看lean大大的 安排了~

uglee, 2018-07-10:
可以 自己比葫芦画瓢试着写个就好。

505575853, 2018-07-21:
https://github.com/505575853/openwrt-ssr.git 用ywb94源码,这个我瞎写的暂时能用

nimasi, 2018-07-24:
@505575853   这个版本订阅功能可以用麻？

nimasi, 2018-07-24:
@505575853  编译gfw模式，缺少dnsforwarder，请问下这个哪里有

505575853, 2018-07-24:
https://github.com/505575853/openwrt-dnsforwarder

nimasi, 2018-07-24:
@505575853  
我改用你的这个dnsforwarder发现，虽然dnsforwarder保存了启用，但是重启后，启用的勾就没了变成未运行了，而且改动ssr的设置也会这样，之前用AlexZhuo的dnsforwarder好像没这种情况

shenzt68, 2018-08-04:
@505575853 这个版本增加了auth_chain_a等协议了吗？

closed 2018-10-06 by coolsnowwolf