coolsnowwolf/lede #473: ./scripts/feeds install -a操作后出现WARNING

MxCen, 2018-07-02:
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'ntfs-3g', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/he-6in4/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/heimdal/Makefile' has a dependency on 'libsqlite3', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/lean/libidn2/Makefile' has a dependency on 'libunistring', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-aliddns/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-aliddns/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-apfree_wifidog/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-brook-pro/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-brook-pro/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/luci-app-brook-pro/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-brook-pro/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-filetransfer/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-flowoffload/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-haproxy-tcp/Makefile' has a dependency on 'haproxy', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-vpnd/Makefile' has a dependency on 'strongswan-minimal', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-vpnd/Makefile' has a dependency on 'strongswan-mod-xauth-generic', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-vpnd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-kcptun/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-kcptun/Makefile' has a dependency on 'luci-lib-jsonc', which does not exist
WARNING: Makefile 'package/lean/luci-app-kcptun/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-mproxy/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-n2n_v2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-ngrokc/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-openvpn-server/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-pptp-server/Makefile' has a dependency on 'pptpd', which does not exist
WARNING: Makefile 'package/lean/luci-app-pptp-server/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-samba4/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-sfe/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-pro/Makefile' has a dependency on 'shadowsocks-libev-ss-redir', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-pro/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-pro/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-pro/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-pro/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssrserver-python/Makefile' has a dependency on 'python', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssrserver-python/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-syncdial/Makefile' has a dependency on 'luci-app-mwan3', which does not exist
WARNING: Makefile 'package/lean/luci-app-syncdial/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-usb-printer/Makefile' has a dependency on 'p910nd', which does not exist
WARNING: Makefile 'package/lean/luci-app-usb-printer/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-v2ray/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-v2ray/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/luci-app-v2ray/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-v2ray/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-vlmcsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-vsftpd/Makefile' has a build dependency on 'luci-base/host', which does not exist
[... 13 lines cut ...]
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libcap', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libtirpc', which does not exist
[... cut ...]
WARNING: Makefile 'package/lean/shadowsocksR-libev-full/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/shadowsocksR-libev-full/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/dvb/tvheadend/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/dvb/tvheadend/Makefile' has a dependency on 'libavahi-client', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist

paldier, 2018-07-19:
应该是update失败,多执行几次确定全部成功了在install吧

thesadboy, 2019-07-29:
遇到了同样的问题，update很多次，也没有失败信息出现，但是只要执行 install 就会提示这样的

phantastar, 2021-01-03:
> 遇到了同样的问题，update很多次，也没有失败信息出现，但是只要执行 install 就会提示这样的

> WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
> WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist [...]

我刚刚各种Google后, 解决了这个问题: 出现这种错误后, 再次执行之前的两条命令时加上"-f"(就是说先./scripts/feeds update -a  -f接着./scripts/feeds install -a -f), 最后重复执行一下./scripts/feeds update -a和./scripts/feeds install -a就修复完毕了.

ymwoo888, 2021-04-30:
我的试了不行