coolsnowwolf/lede #474: ssr问题，去掉“启用DNS TCP防污染”的勾？

yes2003, 2018-07-02:
make menuconfig里只选了x86 64编译无错误
ss账号正常
ssr显示绿色的正在运行，但是打不开墙外网页
把 “启用DNS TCP防污染”的勾去掉倒是能打开
是SSR还是DNS问题？

plutoghost, 2018-07-02:
dns问题，手机客户端也有一样的毛病

MxCen, 2018-07-02:
cp -fpR /home/meng/lede/build_dir/target-mips_24kc_musl/root-ar71xx /home/meng/lede/build_dir/target-mips_24kc_musl/root.orig-ar71xx
/home/meng/lede/build_dir/target-mips_24kc_musl/root-ar71xx/etc/init.d/ssrpro: line 48: network_get_subnet: command not found
/home/meng/lede/build_dir/target-mips_24kc_musl/root-ar71xx/etc/init.d/ssrpro: line 49: network_get_ipaddr: command not found
Create '/home/meng/lede/build_dir/target-mips_24kc_musl/root-ar71xx/etc/filesystems'.
Add 'ntfs-3g' to known filesystems.
./etc/init.d/ssrpro: line 48: network_get_subnet: command not found
./etc/init.d/ssrpro: line 49: network_get_ipaddr: command not found

closed 2018-10-06 by coolsnowwolf