coolsnowwolf/lede #482: LUCI v2ray 没有运行

linlinlin1992, 2018-07-08:
编译了v2ray 和 lua-cjson 进固件，filetransfer也安装了，配置好服务器点击启用，仍然显示：v2ray没有运行。服务器配置设置在电脑上可以运行。ssr相同操作启用成功。

启动项打开，并重启，无效。

查看系统日志，没有与v2ray相关的记录。

ssh 直接使用/usr/bin/v2ray/v2ray -config=/etc/v2ray/config.json
显示Illegal instruction

毫无头绪，somebody help me。

orzsun, 2018-07-09:
下载v2ray完整包放进去

linlinlin1992, 2018-07-09:
@orzsun  不好意思，下载什么包放在哪里？v2ray-linux-mips.zip 吗？放在什么目录呢？

winscp看过了，/usr/bin/v2ray/ 目录是存在的，里面两个文件v2ctl v2ray
/etc/v2ray/ 目录下的配置文件也都在

orzsun, 2018-07-10:
直接覆盖就行,包里少文件了所以没法运行

linlinlin1992, 2018-07-10:
@orzsun zip解压覆盖后依然无法运行

sirfromprc, 2018-07-14:
我是第一次运行v2ray也提示没有运行，不管他，再点击一次确定就可以打开了。

linlinlin1992, 2018-07-14:
@sirfromprc 不知道你说再点击一次确定是怎么做的。
我编译完，输好服务器信息，点击启用，保存并应用。显示红字没有运行。

去启动项把所有服务打开，重启，再改下代理模式，保存并应用，还是没有运行。

直接点保存并应用第二次，会显示没有待应用的更改。

sirfromprc, 2018-07-14:
目前我用的是自己编译的是4.9.102版本，启动项处“v2ray”我没做更改，还是提示禁用状态。你的“直接点保存并应用第二次，会显示没有待应用的更改。”，你应该用的最新luci，前段时间用4.14内核的时候我是第一次保存&应用后再点击“自动更新GFW-List”或其他选项，就可以点“保存&应用”。然后就提示正在运行了了。我ea8500，4.9.102\4.14.42版本均可运行。x64版本我编译的也可以运行。

linlinlin1992, 2018-07-15:
@sirfromprc 我看了下内核版本是	4.9.107。启动项开和不开我都试过了。我选了其他选项，再保存应用仍然不运行。我编译了5-6次吧，还试过brook，就是不运行，brook也不运行……只有SS（r）能跑。
我是真的没招了，哪怕哪里有个log能看看出了什么错也好，我也有个方向……

505575853, 2018-07-21:
define Package/$(PKG_NAME)/config
menu "V2Ray Configuration"
	depends on PACKAGE_v2ray

config PACKAGE_V2RAY_INCLUDE_V2RAY
	bool "Include v2ray"
	default y

config PACKAGE_V2RAY_SOFTFLOAT
	bool "Use soft-float binaries (mips/mipsle only)"
	depends on PACKAGE_V2RAY_INCLUDE_V2RAY
	default n

config PACKAGE_V2RAY_INCLUDE_V2CTL
	bool "Include v2ctl"
	depends on PACKAGE_V2RAY_INCLUDE_V2RAY
	default y

config PACKAGE_V2RAY_INCLUDE_GEOIP
	bool "Include geoip.dat"
	depends on PACKAGE_V2RAY_INCLUDE_V2CTL
#	default n
	default y

config PACKAGE_V2RAY_INCLUDE_GEOSITE
	bool "Include geosite.dat"
	depends on PACKAGE_V2RAY_INCLUDE_V2CTL
#	default n
	default y

endmenu
endef

有两个文件默认是没选上的

505575853, 2018-07-21:
我可以显示运行就是无法翻墙,估计配置文件哪里不对不知道有没大佬教下怎么生成配置

linlinlin1992, 2018-07-26:
@505575853 这几个都编译进去了还是不行。

你配置文件放在别的客户端上能成功翻么

505575853, 2018-07-26:
你看原作者说的暂时只能用手工填写,直接传配置文件运行不了,我手工填写运行成功并能翻出去了

linlinlin1992, 2018-07-28:
@505575853 昨天刚好重装了下系统，晚上重新编译完，今天刷完就能用了 orz

绝了，这编译还能出什么问题……一模一样的命令啊……

closed 2018-07-28 by linlinlin1992

iotsys, 2019-01-05:
是不是虚拟FPU没有开启？不过，我记得已经改成默认开启了

python777, 2019-12-06:
3700v4. 也有可能是存储容量不够导致的。v2ray最小也有9M，但路由器只有8M