coolsnowwolf/lede #484: 编译出错求助（log）

azazjjyjjy, 2018-07-14:
试了好几次都是卡在这里，全局代理也加了。请问是什么原因
ipq806x-r7800-v2ray 其他设置没改

checking whether mknod can create fifo without root privileges... configure: error: in `/home/guo/LEDE2/lede/build_dir/host/tar-1.30':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make[3]: *** [/home/guo/LEDE2/lede/build_dir/host/tar-1.30/.configured] Error 1
make[3]: Leaving directory `/home/guo/LEDE2/lede/tools/tar'
make[2]: *** [tools/tar/compile] Error 2
make[2]: Leaving directory `/home/guo/LEDE2/lede'
make[1]: *** [/home/guo/LEDE2/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.tools_compile_yynyyyyynyyyyynyynnyyyynyyyyyyyyyyyyyyynyynynnyyynny] Error 2
make[1]: Leaving directory `/home/guo/LEDE2/lede'
make: *** [world] 错误 2

sirfromprc, 2018-07-14:
我ea8500、x64编译这个都没有问题。“/home/guo/LEDE2/lede/build_dir/host/tar-1.30': configure: error: you should not run configure as root“看提示好像是不要以root用户编译。

azazjjyjjy, 2018-07-14:
编译前运行set FORCE_UNSAFE_CONFIGURE=1 解决

closed 2018-07-14 by azazjjyjjy