coolsnowwolf/lede #489: 编译生成不了bin固件

jzhou404, 2018-07-22:
测试了7620的Y1S选项，和7621的wr1200js，都是同样出错，报错是： is too big！谢谢。 log如下：

[ -f /home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/youhua_wr1200js-kernel.bin -a -f /home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/root.squashfs ]
dd if=/home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/youhua_wr1200js-kernel.bin >> /home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin
3480+1 records in
3480+1 records out
1782002 bytes (1.8 MB) copied, 0.0110086 s, 162 MB/s
dd if=/home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/root.squashfs >> /home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin
32916+1 records in
32916+1 records out
16853384 bytes (17 MB) copied, 0.0552411 s, 305 MB/s
/home/jzhou/lede/staging_dir/host/bin/padjffs2 /home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin  4 8 16 64 128 256
padding image to 011c6000
padding image to 011c8000
padding image to 011d0000
padding image to 011e0000
padding image to 01200000
echo '{  "supported_devices":["youhua,wr1200js"], "version": { "dist": "OpenWrt", "version": "SNAPSHOT", "revision": "r565-db75f73", "board": "ramips" } }' | fwtool -I - /home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin
WARNING: Image file /home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin is too big
cp /home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin /home/jzhou/lede/bin/targets/ramips/mt7621/openwrt-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin
cp: cannot stat '/home/jzhou/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin': No such file or directory
make[5]: [/home/jzhou/lede/bin/targets/ramips/mt7621/openwrt-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin] Error 1 (ignored)
IPKG_NO_SCRIPT=1 IPKG_INSTROOT=/home/jzhou/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips TMPDIR=/home/jzhou/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips/tmp /home/jzhou/lede/staging_dir/host/bin/opkg --offline-root /home/jzhou/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch mipsel_24kc:200 list-installed > /home/jzhou/lede/bin/targets/ramips/mt7621/openwrt-ramips-mt7621-device-youhua-wr1200js.manifest

pk49800, 2018-07-29:
好像说是samba的问题，你在luci菜单里面把和samba有关的去掉再试试？

closed 2018-10-06 by coolsnowwolf

ZhuangJW, 2018-10-13:
怎么修改FLASH的大小？