coolsnowwolf/lede #493: Openvpn连接上后报错没网

SUNJINZE, 2018-08-01:
Wed Aug 01 13:46:30 2018 OpenVPN 2.4.6 x86_64-w64-mingw32 [SSL (OpenSSL)] [LZO] [LZ4] [PKCS11] [AEAD] built on Apr 26 2018
Wed Aug 01 13:46:30 2018 Windows version 6.2 (Windows 8 or greater) 64bit
Wed Aug 01 13:46:30 2018 library versions: OpenSSL 1.1.0h  27 Mar 2018, LZO 2.10
Wed Aug 01 13:46:30 2018 MANAGEMENT: TCP Socket listening on [AF_INET]127.0.0.1:25340
Wed Aug 01 13:46:30 2018 Need hold release from management interface, waiting...
Wed Aug 01 13:46:30 2018 MANAGEMENT: Client connected from [AF_INET]127.0.0.1:25340
Wed Aug 01 13:46:30 2018 MANAGEMENT: CMD 'state on'
Wed Aug 01 13:46:30 2018 MANAGEMENT: CMD 'log all on'
Wed Aug 01 13:46:30 2018 MANAGEMENT: CMD 'echo all on'
Wed Aug 01 13:46:30 2018 MANAGEMENT: CMD 'bytecount 5'
Wed Aug 01 13:46:30 2018 MANAGEMENT: CMD 'hold off'
Wed Aug 01 13:46:30 2018 MANAGEMENT: CMD 'hold release'
### Wed Aug 01 13:46:30 2018 WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Wed Aug 01 13:46:30 2018 MANAGEMENT: >STATE:1533102390,RESOLVE,,,,,,
Wed Aug 01 13:46:31 2018 TCP/UDP: Preserving recently used remote address: [AF_INET]115.218.103.13:1194
Wed Aug 01 13:46:31 2018 Socket Buffers: R=[65536->65536] S=[65536->65536]
Wed Aug 01 13:46:31 2018 Attempting to establish TCP connection with [AF_INET]115.218.103.13:1194 [nonblock]
Wed Aug 01 13:46:31 2018 MANAGEMENT: >STATE:1533102391,TCP_CONNECT,,,,,,
Wed Aug 01 13:46:32 2018 TCP connection established with [AF_INET]115.218.103.13:1194
Wed Aug 01 13:46:32 2018 TCP_CLIENT link local: (not bound)
Wed Aug 01 13:46:32 2018 TCP_CLIENT link remote: [AF_INET]115.218.103.13:1194
Wed Aug 01 13:46:32 2018 MANAGEMENT: >STATE:1533102392,WAIT,,,,,,
Wed Aug 01 13:46:32 2018 MANAGEMENT: >STATE:1533102392,AUTH,,,,,,
Wed Aug 01 13:46:32 2018 TLS: Initial packet from [AF_INET]115.218.103.13:1194, sid=bdb87042 06667ced
Wed Aug 01 13:46:32 2018 VERIFY OK: depth=1, C=CN, ST=ZJ, L=ZJ, O=ZJ, OU=ZJ, CN=ZJ CA, name=EasyRSA, emailAddress=ZJ@ZJ.com
Wed Aug 01 13:46:32 2018 VERIFY OK: depth=0, C=CN, ST=ZJ, L=ZJ, O=ZJ, OU=ZJ, CN=server, name=EasyRSA, emailAddress=ZJ@ZJ.com
Wed Aug 01 13:46:33 2018 Control Channel: TLSv1.2, cipher TLSv1.2 ECDHE-RSA-AES256-GCM-SHA384, 1024 bit RSA
Wed Aug 01 13:46:33 2018 [server] Peer Connection Initiated with [AF_INET]115.218.103.13:1194
Wed Aug 01 13:46:34 2018 MANAGEMENT: >STATE:1533102394,GET_CONFIG,,,,,,
Wed Aug 01 13:46:34 2018 SENT CONTROL [server]: 'PUSH_REQUEST' (status=1)
Wed Aug 01 13:46:34 2018 PUSH: Received control message: 'PUSH_REPLY,route 10.0.0.1 255.255.255.0,comp-lzo adaptive,redirect-gateway def1 bypass-dhcp,dhcp-option DNS 10.0.0.1,route-gateway 10.8.0.1,topology subnet,ping 10,ping-restart 120,ifconfig 10.8.0.2 255.255.255.0,peer-id 0,cipher AES-256-GCM'
Wed Aug 01 13:46:34 2018 OPTIONS IMPORT: timers and/or timeouts modified
Wed Aug 01 13:46:34 2018 OPTIONS IMPORT: compression parms modified
Wed Aug 01 13:46:34 2018 OPTIONS IMPORT: --ifconfig/up options modified
Wed Aug 01 13:46:34 2018 OPTIONS IMPORT: route options modified
Wed Aug 01 13:46:34 2018 OPTIONS IMPORT: route-related options modified
Wed Aug 01 13:46:34 2018 OPTIONS IMPORT: --ip-win32 and/or --dhcp-option options modified
Wed Aug 01 13:46:34 2018 OPTIONS IMPORT: peer-id set
Wed Aug 01 13:46:34 2018 OPTIONS IMPORT: adjusting link_mtu to 1626
Wed Aug 01 13:46:34 2018 OPTIONS IMPORT: data channel crypto options modified
Wed Aug 01 13:46:34 2018 Data Channel: using negotiated cipher 'AES-256-GCM'
Wed Aug 01 13:46:34 2018 Outgoing Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
Wed Aug 01 13:46:34 2018 Incoming Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
Wed Aug 01 13:46:34 2018 interactive service msg_channel=0
Wed Aug 01 13:46:34 2018 ROUTE_GATEWAY 192.168.1.1/255.255.255.0 I=12 HWADDR=f4:6d:04:ce:fb:9a
Wed Aug 01 13:46:34 2018 open_tun
Wed Aug 01 13:46:34 2018 TAP-WIN32 device [Npcap Loopback Adapter] opened: \\.\Global\{C49E5FF1-0ED0-493E-9A1E-AB069152DD56}.tap
Wed Aug 01 13:46:34 2018 TAP-Windows Driver Version 9.21 
Wed Aug 01 13:46:34 2018 Set TAP-Windows TUN subnet mode network/local/netmask = 10.8.0.0/10.8.0.2/255.255.255.0 [SUCCEEDED]
Wed Aug 01 13:46:34 2018 Notified TAP-Windows driver to set a DHCP IP/netmask of 10.8.0.2/255.255.255.0 on interface {C49E5FF1-0ED0-493E-9A1E-AB069152DD56} [DHCP-serv: 10.8.0.254, lease-time: 31536000]
Wed Aug 01 13:46:34 2018 Successful ARP Flush on interface [11] {C49E5FF1-0ED0-493E-9A1E-AB069152DD56}
Wed Aug 01 13:46:34 2018 do_ifconfig, tt->did_ifconfig_ipv6_setup=0
Wed Aug 01 13:46:34 2018 MANAGEMENT: >STATE:1533102394,ASSIGN_IP,,10.8.0.2,,,,
Wed Aug 01 13:46:39 2018 TEST ROUTES: 2/2 succeeded len=1 ret=1 a=0 u/d=up
Wed Aug 01 13:46:39 2018 C:\WINDOWS\system32\route.exe ADD 115.218.103.13 MASK 255.255.255.255 192.168.1.1
Wed Aug 01 13:46:39 2018 ROUTE: CreateIpForwardEntry succeeded with dwForwardMetric1=35 and dwForwardType=4
Wed Aug 01 13:46:39 2018 Route addition via IPAPI succeeded [adaptive]
Wed Aug 01 13:46:39 2018 C:\WINDOWS\system32\route.exe ADD 0.0.0.0 MASK 128.0.0.0 10.8.0.1
Wed Aug 01 13:46:39 2018 ROUTE: CreateIpForwardEntry succeeded with dwForwardMetric1=35 and dwForwardType=4
Wed Aug 01 13:46:39 2018 Route addition via IPAPI succeeded [adaptive]
Wed Aug 01 13:46:39 2018 C:\WINDOWS\system32\route.exe ADD 128.0.0.0 MASK 128.0.0.0 10.8.0.1
Wed Aug 01 13:46:40 2018 ROUTE: CreateIpForwardEntry succeeded with dwForwardMetric1=35 and dwForwardType=4
Wed Aug 01 13:46:40 2018 Route addition via IPAPI succeeded [adaptive]
Wed Aug 01 13:46:40 2018 MANAGEMENT: >STATE:1533102400,ADD_ROUTES,,,,,,
Wed Aug 01 13:46:40 2018 C:\WINDOWS\system32\route.exe ADD 10.0.0.1 MASK 255.255.255.0 10.8.0.1
### Wed Aug 01 13:46:40 2018 Warning: address 10.0.0.1 is not a network address in relation to netmask 255.255.255.0
### Wed Aug 01 13:46:40 2018 ROUTE: route addition failed using CreateIpForwardEntry: ²ÎÊݴíÎ㡣   [status=87 if_index=11]
Wed Aug 01 13:46:40 2018 Route addition via IPAPI failed [adaptive]
Wed Aug 01 13:46:40 2018 Route addition fallback to route.exe
Wed Aug 01 13:46:40 2018 env_block: add PATH=C:\WINDOWS\System32;C:\WINDOWS;C:\WINDOWS\System32\Wbem
### Wed Aug 01 13:46:40 2018 WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Wed Aug 01 13:46:40 2018 Initialization Sequence Completed
Wed Aug 01 13:46:40 2018 MANAGEMENT: >STATE:1533102400,CONNECTED,SUCCESS,10.8.0.2,115.218.103.13,1194,192.168.1.153,62604
### Wed Aug 01 13:46:43 2018 Bad LZO decompression header byte: 42
Wed Aug 01 13:46:53 2018 Bad LZO decompression header byte: 42
Wed Aug 01 13:47:03 2018 Bad LZO decompression header byte: 42
Wed Aug 01 13:47:13 2018 Bad LZO decompression header byte: 42
Wed Aug 01 13:47:23 2018 Bad LZO decompression header byte: 42
Wed Aug 01 13:47:33 2018 Bad LZO decompression header byte: 42
Wed Aug 01 13:47:43 2018 Bad LZO decompression header byte: 42
Wed Aug 01 13:47:54 2018 Bad LZO decompression header byte: 42
Wed Aug 01 13:48:04 2018 Bad LZO decompression header byte: 42

closed 2018-08-07 by SUNJINZE

honnbin, 2018-08-11:
我也遇到这个问题了，老版本没有问题，新固件连接上了但是上不去网

honnbin, 2018-08-11:
最后怎么解决的啊？

SUNJINZE, 2018-08-11:
使用老版本客户端即可

honnbin, 2018-08-11:
确实，我用2.3.x版本的就能连接，2.4.x就出错，thx