coolsnowwolf/lede #504: 最新代码编译acrh17时，openwrt-ipq40xx-asus_rt-acrh17-initramfs-factory.trx.new不存在

523860169, 2018-08-11:
```
/home/debian/lede/staging_dir/host/bin/lzma e /home/debian/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-asus_rt-acrh17-initramfs-factory.trx -lc1 -lp2 -pb2  /home/debian/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-asus_rt-acrh17-initramfs-factory.trx.new
mv: cannot stat '/home/debian/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-asus_rt-acrh17-initramfs-factory.trx.new': No such file or directory
Makefile:227: recipe for target '/home/debian/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-asus_rt-acrh17-initramfs-factory.trx' failed
make[5]: *** [/home/debian/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-asus_rt-acrh17-initramfs-factory.trx] Error 1
make[5]: Leaving directory '/home/debian/lede/target/linux/ipq40xx/image'
Makefile:22: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/debian/lede/target/linux/ipq40xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/debian/lede/target/linux'
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/debian/lede'
target/Makefile:19: recipe for target '/home/debian/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install' failed
make[1]: *** [/home/debian/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/debian/lede'
/home/debian/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
debian@debian:~/lede$ 
```

closed 2018-08-15 by 523860169

forsan-co, 2018-12-21:
@523860169  您好，请问 target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install' failed 这个问题有解决吗？我也遇到了

523860169, 2018-12-21:
@forsan-co 太久了，忘记怎么解决的了