coolsnowwolf/lede #514: SSR似乎有问题

450717565, 2018-08-24:
最近经常发现上不了外网，看节点又是正常的，需要重新点一下 手动强制更新GFW-List 这个就会正常，我是搭配 HAProxy 用的，不知道是什么情况？

coolsnowwolf, 2018-08-24:
用新版本看看，估计是 dnsmasq 问题

450717565, 2018-08-27:
[2018-08-28 02:00:00] ShadowsocksR No Problem.
[2018-08-28 01:00:00] ShadowsocksR No Problem.
udhcpc: no lease, failing
udhcpc: sending discover
udhcpc: started, v1.28.3
udhcpc: no lease, failing
udhcpc: sending discover
udhcpc: started, v1.28.3
[2018-08-28 00:00:00] Problem decteted, restarting ShadowsocksR...
[2018-08-27 23:00:00] ShadowsocksR No Problem.

liganbo, 2018-08-28:
udhcpc: no lease, failing
udhcpc: sending discover
udhcpc: started, v1.28.3
 2018-08-28 22:00:06 INFO: obfs_param 
 2018-08-28 22:00:06 INFO: obfs plain
 2018-08-28 22:00:06 INFO: method rc4-md5
 2018-08-28 22:00:06 INFO: protocol_param 
 2018-08-28 22:00:06 INFO: protocol origin
udhcpc: no lease, failing
udhcpc: sending discover
udhcpc: started, v1.28.3
[2018-08-28 22:00:00] Problem decteted, restarting ShadowsocksR...
[2018-08-28 21:00:00] ShadowsocksR No Problem.
[2018-08-28 20:00:00] ShadowsocksR No Problem.

同样出现这个

closed 2018-10-06 by coolsnowwolf