coolsnowwolf/lede #519: ar71xx生成固件时报错,[search space used is default of 16KB] [Command line marker not found! ]

nimasi, 2018-08-25:
`Ignoring, exclude it (-e/-ef) to override.
[========================================================================================|] 991/991 100%
Exportable Squashfs 4.0 filesystem, xz compressed, data block size 1048576
	compressed data, compressed metadata, compressed fragments, no xattrs
	duplicates are removed
Filesystem size 2401.98 Kbytes (2.35 Mbytes)
	30.58% of uncompressed filesystem size (7855.53 Kbytes)
Inode table size 8932 bytes (8.72 Kbytes)
	21.09% of uncompressed inode table size (42354 bytes)
Directory table size 12956 bytes (12.65 Kbytes)
	47.12% of uncompressed directory table size (27495 bytes)
Number of duplicate files found 114
Number of inodes 1293
Number of files 996
Number of fragments 8
Number of symbolic links  185
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 111
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
rm -f /media/x/3/lede-master/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tl-wr741nd-v4-kernel.bin
cp /media/x/3/lede-master/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/vmlinux /media/x/3/lede-master/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tl-wr741nd-v4-kernel.bin
/media/x/3/lede-master/staging_dir/host/bin/patch-cmdline /media/x/3/lede-master/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tl-wr741nd-v4-kernel.bin 'board=TL-WR741ND-v4  console=ttyATH0,115200'
search space used is default of 16KB
Command line marker not found!
Makefile:68: recipe for target '/media/x/3/lede-master/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tl-wr741nd-v4-kernel.bin' failed
make[5]: *** [/media/x/3/lede-master/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tl-wr741nd-v4-kernel.bin] Error 255
make[5]: Leaving directory '/media/x/3/lede-master/target/linux/ar71xx/image'
Makefile:24: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/media/x/3/lede-master/target/linux/ar71xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/media/x/3/lede-master/target/linux'
Command exited with non-zero status 2
time: target/linux/install#12.09#3.05#13.57
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/media/x/3/lede-master'
target/Makefile:19: recipe for target '/media/x/3/lede-master/staging_dir/target-mips_24kc_musl/stamp/.target_install' failed
make[1]: *** [/media/x/3/lede-master/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/media/x/3/lede-master'
/media/x/3/lede-master/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
`

nimasi, 2018-08-25:
> search space used is default of 16KB
> Command line marker not found!
> Makefile:68: recipe for target '/media/x/3/lede-master/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tl-wr741nd-v4-kernel.bin' failed

没有修改任何文件 ,只保留了基础软件包,是否是最近的commit对rom和image大小修改导致的?

nimasi, 2018-08-26:
是同步到18.06.1导致的无法切换4.14内核

nimasi, 2018-08-26:
有人指出同步错了,同步的是18.06 ,不是18.06.1

https://github.com/coolsnowwolf/lede/commit/9ba04fd0d795a96cf5fb00a997a4885b82bbf529#r30265646

coolsnowwolf, 2018-08-26:
没有同步错，是 18.06.1 branch

nimasi, 2018-08-26:
重新克隆的仓库，没有编译过，跟dirclean没有关系

SouHong, 2018-08-27:
没有/target/linux/ar71xx/patches-4.14能否正常编译？

coolsnowwolf, 2018-08-27:
官方 ar71xx 不支持 4.14 内核

closed 2018-10-06 by coolsnowwolf