coolsnowwolf/lede #522: 大雕你好，OpenWrt-18.06.1下，luci-app-zerotier “启用” 选项无效。

hzqim, 2018-08-28:
必须通过 /etc/init.d/zerotier 命令来启停。
谢谢。

fuqiang03, 2018-08-30:
OpenWrt-18.06 不光这个一个插件哦  部分插件都受18.06的luci影响

hzqim, 2018-08-30:
@fuqiang03 是吗？
目前我只碰到这个luci有问题。

fuqiang03, 2018-08-30:
@hzqim 插件安装的太少了，像我这样就可以了
[image]
[image]
[image]
[image]

closed 2018-10-06 by coolsnowwolf

arfaWong, 2019-03-26:
@fuqiang03 你图里面的光猫访问插件是哪个，可以分享下吗？

fuqiang03, 2019-03-26:
@arfaWong 
https://github.com/openwrt-develop/openwrt/tree/master/package/mypackages/luci-app-modem-access