coolsnowwolf/lede #525: wndr4300v1 fast-path 编译error：'struct sk_buff' has no member named 'fast_forwarded'   skb->fast_forwarded = 1;

Lancenas, 2018-08-29:
make[4]: Entering directory '/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/linux-4.4.151'
  CC [M]  /home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe/sfe_ipv4.o
/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_udp':
/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe/sfe_ipv4.c:1366:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^
/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_tcp':
/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe/sfe_ipv4.c:1909:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^
scripts/Makefile.build:269: recipe for target '/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe/sfe_ipv4.o' failed
make[5]: *** [/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe/sfe_ipv4.o] Error 1
Makefile:1421: recipe for target '_module_/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe' failed
make[4]: *** [_module_/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe] Error 2
make[4]: Leaving directory '/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/linux-4.4.151'
Makefile:117: recipe for target '/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe/.built' failed
make[3]: *** [/home/lixiao/lede/source/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_nand/shortcut-fe/.built] Error 2
make[3]: Leaving directory '/home/lixiao/lede/source/package/lean/shortcut-fe'
package/Makefile:105: recipe for target 'package/lean/shortcut-fe/compile' failed
make[2]: *** [package/lean/shortcut-fe/compile] Error 2
make[2]: Leaving directory '/home/lixiao/lede/source'
package/Makefile:101: recipe for target '/home/lixiao/lede/source/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/lixiao/lede/source/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lixiao/lede/source'
/home/lixiao/lede/source/include/toplevel.mk:205: recipe for target 'world' failed
make: *** [world] Error 2

closed 2018-10-06 by coolsnowwolf

hzjnet, 2019-01-14:
@Lancenas 我也遇见类似问题 请教具体如何处理的喃？ 谢谢

jsda, 2019-01-14:
编译4.19的内核也遇到此问题，水平不够，没看懂上面那篇文章是怎么解决的。
@hzjnet 朋友解决了么？