coolsnowwolf/lede #528: Makefile:2278: recipe for target 's-automata' failed make[5]: *** [s-automata] Error 137

Lancenas, 2018-08-30:
全部按步骤编译出现以下错误：
/bin/bash: line 1: 24342 Killed                  build/genautomata /home/lixiao/lede/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/common.md /home/lixiao/lede/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/config/mips/mips.md insn-conditions.md > tmp-automata.c
Makefile:2278: recipe for target 's-automata' failed
make[5]: *** [s-automata] Error 137
make[5]: Leaving directory '/home/lixiao/lede/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial/gcc'
Makefile:4210: recipe for target 'all-gcc' failed
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory '/home/lixiao/lede/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial'
Makefile:36: recipe for target '/home/lixiao/lede/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial/.built' failed
make[3]: *** [/home/lixiao/lede/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial/.built] Error 2
make[3]: Leaving directory '/home/lixiao/lede/lede/toolchain/gcc/initial'
Command exited with non-zero status 2
time: toolchain/gcc/initial/compile#225.06#44.18#351.68
toolchain/Makefile:96: recipe for target 'toolchain/gcc/initial/compile' failed
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/lixiao/lede/lede'
toolchain/Makefile:94: recipe for target '/home/lixiao/lede/lede/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/lixiao/lede/lede/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/lixiao/lede/lede'
/home/lixiao/lede/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

LGA1150, 2018-08-31:
> /bin/bash: line 1: 24342 Killed

OOM 了
这一步会占用至少 1.5G RAM，如果在虚拟机或 VPS 里编译，把 swap 开大点

Lancenas, 2018-09-01:
LGA1150 确实是虚拟机内存小了，谢谢

closed 2018-10-06 by coolsnowwolf

tufu9441, 2019-12-11:
终于解决问题了，感谢！