coolsnowwolf/lede #530: 编译刷机正常，18.06 v2ray无法启动？

Lancenas, 2018-08-30:
BusyBox v1.28.3 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 18.06-SNAPSHOT, r583-b8577d51
 -----------------------------------------------------
root@OpenWrt:~# /etc/init.d/v2raypro start
WARNING: V2Ray is disabled.
root@OpenWrt:~#

lyin888, 2018-08-30:
lua-cjson 依赖包有没有安装

Lancenas, 2018-08-30:
lua-cjson安装了，ss禁用了

lyin888, 2018-08-30:
系统-启动项 重启 v2ray  试试

Lancenas, 2018-08-30:
重启没用，另外守护日志没信息

lyin888, 2018-09-01:
估计编译不完整

honnbin, 2018-09-01:
除了lua-cjson，还要选择luci-filetransfer什么的

Lancenas, 2018-09-01:
支持包都安装了，感觉是权限问题，终端启动提示非法引导，没发现怎么解决

lyin888, 2018-09-02:
你路由型号，CPU型号，应该是你v2ray包编译不完整

lyin888, 2018-09-02:
[image]

路由 水星MW4530R ，CPU AR9344 
MIPS CPU 编译的时候开启fpu浮点
还有等等暂时记不起来了

lyin888, 2018-09-02:
v2ray_v3.26_mips_24kc.ipk
完整包有 8.22 MB (8,621,448 字节)

XavierPENROSE, 2018-09-13:
楼主问题解决了吗

closed 2018-10-06 by coolsnowwolf