coolsnowwolf/lede #538: BBR是否能正常编译？

SouHong, 2018-09-10:
/home/zs/ledelean/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/bbr_mod/tcp_bbr_mod.c:863:2: note: in expansion of macro 'cmpxchg'
  cmpxchg(&sk->sk_pacing_status, SK_PACING_NONE, SK_PACING_NEEDED);
  ^~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:302: recipe for target '/home/zs/ledelean/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/bbr_mod/tcp_bbr_mod.o' failed
make[5]: *** [/home/zs/ledelean/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/bbr_mod/tcp_bbr_mod.o] Error 1
Makefile:1508: recipe for target '_module_/home/zs/ledelean/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/bbr_mod' failed
make[4]: *** [_module_/home/zs/ledelean/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/bbr_mod] Error 2
make[4]: Leaving directory '/home/zs/ledelean/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.124'
Makefile:46: recipe for target '/home/zs/ledelean/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/bbr_mod/.built' failed
make[3]: *** [/home/zs/ledelean/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/bbr_mod/.built] Error 2
make[3]: Leaving directory '/home/zs/ledelean/package/lean/bbr'
time: package/lean/bbr/compile#0.74#0.98#1.84
package/Makefile:107: recipe for target 'package/lean/bbr/compile' failed
make[2]: *** [package/lean/bbr/compile] Error 2
make[2]: Leaving directory '/home/zs/ledelean'
package/Makefile:103: recipe for target '/home/zs/ledelean/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/zs/ledelean/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zs/ledelean'
/home/zs/ledelean/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

hcym, 2018-09-10:
能编译，就是友华的开硬件加速后看不到bbr，ac58u开软件加速后还是可以看到bbr模块正常使用，非要用的话官方分支的感觉靠谱一点，其实没什么效果

hcym, 2018-09-11:
k2的只有软件加速，开了也正常，看来和什么offload有关

closed 2018-09-12 by SouHong