coolsnowwolf/lede #546: 14.04 ltsmake download出错

chaihuo, 2018-09-13:
报错1：
make[2]: Entering directory `/home/bl/netgear/lede/package/lean/n2n_v2'
mkdir -p /home/bl/netgear/lede/dl
SHELL= flock /home/bl/netgear/lede/tmp/.n2n_v2-9349.tar.gz.flock -c '    echo "Checking out files from the svn repository..."; mkdir -p /home/bl/netgear/lede/tmp/dl && cd /home/bl/netgear/lede/tmp/dl && rm -rf n2n_v2-9349 && [ \! -d n2n_v2-9349 ] && ( svn help export | grep -q trust-server-cert && svn export --non-interactive --trust-server-cert -r9349 https://shop.ntop.org/svn/ntop/trunk/n2n/n2n_v2 n2n_v2-9349 || svn export --non-interactive -r9349 https://shop.ntop.org/svn/ntop/trunk/n2n/n2n_v2 n2n_v2-9349 ) && echo "Packing checkout..." && export TAR_TIMESTAMP="" &&         tar --numeric-owner --owner=0 --group=0 --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c n2n_v2-9349 |         gzip -nc > /home/bl/netgear/lede/tmp/dl/n2n_v2-9349.tar.gz && mv /home/bl/netgear/lede/tmp/dl/n2n_v2-9349.tar.gz /home/bl/netgear/lede/dl/ && rm -rf n2n_v2-9349;     '
Checking out files from the svn repository...
svn: E175011: Unable to connect to a repository at URL 'https://shop.ntop.org/svn/ntop/trunk/n2n/n2n_v2'
svn: E175011: Repository moved temporarily to 'https://svn.ntop.org/svn/ntop/trunk/n2n/n2n_v2'; please relocate
svn: E175011: Unable to connect to a repository at URL 'https://shop.ntop.org/svn/ntop/trunk/n2n/n2n_v2'
svn: E175011: Repository moved temporarily to 'https://svn.ntop.org/svn/ntop/trunk/n2n/n2n_v2'; please relocate
make[2]: *** [/home/bl/netgear/lede/dl/n2n_v2-9349.tar.gz] Error 1
make[2]: Leaving directory `/home/bl/netgear/lede/package/lean/n2n_v2'
time: package/lean/n2n_v2/download#0.14#0.05#5.00
make[1]: *** [package/lean/n2n_v2/download] Error 2

报错2：

time: package/feeds/packages/twisted/python/download#3.78#0.62#384.64
svn: E175011: Unable to connect to a repository at URL 'https://shop.ntop.org/svn/ntop/trunk/n2n/n2n_v2'
svn: E175011: Repository moved temporarily to 'https://svn.ntop.org/svn/ntop/trunk/n2n/n2n_v2'; please relocate
make[2]: time: package/feeds/packages/ocserv/download#3.44#0.54#395.57
*** [/home/bl/netgear/lede/dl/n2n_v2-9349.tar.gz] Error 1
make[2]: Leaving directory `/home/bl/netgear/lede/package/lean/n2n_v2'

报错3：
SHELL= flock /home/bl/netgear/lede/tmp/.n2n_v2-9349.tar.gz.flock -c '    echo "Checking out files from the svn repository..."; mkdir -p /home/bl/netgear/lede/tmp/dl && cd /home/bl/netgear/lede/tmp/dl && rm -rf n2n_v2-9349 && [ \! -d n2n_v2-9349 ] && ( svn help export | grep -q trust-server-cert && svn export --non-interactive --trust-server-cert -r9349 https://shop.ntop.org/svn/ntop/trunk/n2n/n2n_v2 n2n_v2-9349 || svn export --non-interactive -r9349 https://shop.ntop.org/svn/ntop/trunk/n2n/n2n_v2 n2n_v2-9349 ) && echo "Packing checkout..." && export TAR_TIMESTAMP="" &&         tar --numeric-owner --owner=0 --group=0 --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c n2n_v2-9349 |         gzip -nc > /home/bl/netgear/lede/tmp/dl/n2n_v2-9349.tar.gz && mv /home/bl/netgear/lede/tmp/dl/n2n_v2-9349.tar.gz /home/bl/netgear/lede/dl/ && rm -rf n2n_v2-9349;     '
Checking out files from the svn repository...
svn: E175011: Unable to connect to a repository at URL 'https://shop.ntop.org/svn/ntop/trunk/n2n/n2n_v2'
svn: E175011: Repository moved temporarily to 'https://svn.ntop.org/svn/ntop/trunk/n2n/n2n_v2'; please relocate
svn: E175011: Unable to connect to a repository at URL 'https://shop.ntop.org/svn/ntop/trunk/n2n/n2n_v2'
svn: E175011: Repository moved temporarily to 'https://svn.ntop.org/svn/ntop/trunk/n2n/n2n_v2'; please relocate
make[2]: *** [/home/bl/netgear/lede/dl/n2n_v2-9349.tar.gz] Error 1
make[2]: Leaving directory `/home/bl/netgear/lede/package/lean/n2n_v2'
time: package/lean/n2n_v2/download#0.14#0.05#5.00
make[1]: *** [package/lean/n2n_v2/download] Error 2

shijiancat, 2018-09-14:
n2n的源码下载地址失效了 留下邮箱给你发个文件 解压了自己丢进 pakeage - lean 文件夹替换就好

chaihuo, 2018-09-14:
461668765@qq.com感谢哈

chaihuo, 2018-09-14:
> n2n的源码下载地址失效了 留下邮箱给你发个文件 解压了自己丢进 pakeage - lean 文件夹替换就好

461668765@qq.com感谢哈

zhg-em, 2018-09-19:
> n2n的源码下载地址失效了 留下邮箱给你发个文件 解压了自己丢进 pakeage - lean 文件夹替换就好

能不能发给我一份呢？这里先谢过！zhg_em@qq.com

chaihuo, 2018-09-20:
> n2n的源码下载地址失效了 留下邮箱给你发个文件 解压了自己丢进 pakeage - lean 文件夹替换就好

hi，又有新的报错了

/bin/bash: init.l-tmp: No such file or directory
Makefile:1863: recipe for target 'init.c' failed
make[6]: *** [init.c] Error 1
make[6]: Leaving directory '/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1/gdb'
Makefile:8792: recipe for target 'all-gdb' failed
make[5]: *** [all-gdb] Error 2
make[5]: Leaving directory '/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1'
Makefile:849: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1'
Makefile:86: recipe for target '/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1/.built' failed
make[3]: *** [/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1/.built] Error 2
make[3]: Leaving directory '/home/bl/netgear/lede/package/devel/gdb'
time: package/devel/gdb/compile#669.72#197.91#830.40
package/Makefile:107: recipe for target 'package/devel/gdb/compile' failed
make[2]: *** [package/devel/gdb/compile] Error 2
make[2]: Leaving directory '/home/bl/netgear/lede'
package/Makefile:103: recipe for target '/home/bl/netgear/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/bl/netgear/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bl/netgear/lede'
/home/bl/netgear/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
~

zhg-em, 2018-09-20:
> hi，又有新的报错了
把Development->gdb去掉，这个在路由上并不需要，是一个调试器而已！

chaihuo, 2018-09-20:
> > hi，又有新的报错了
> > 把Development->gdb去掉，这个在路由上并不需要，是一个调试器而已！

好的，谢啦！

chaihuo, 2018-09-21:
> > > hi，又有新的报错了
> > > 把Development->gdb去掉，这个在路由上并不需要，是一个调试器而已！
> 
> 好的，谢啦！

> > hi，又有新的报错了
> > 把Development->gdb去掉，这个在路由上并不需要，是一个调试器而已！

您好，去掉了gdb又有新的报错。请问您有可用的.config文件不。
Makefile:1624: recipe for target 'init.o' failed
make[6]: *** [init.o] Error 1
make[6]: Leaving directory '/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1/gdb'
Makefile:8792: recipe for target 'all-gdb' failed
make[5]: *** [all-gdb] Error 2
make[5]: Leaving directory '/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1'
Makefile:849: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1'
Makefile:86: recipe for target '/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1/.built' failed
make[3]: *** [/home/bl/netgear/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gdb-8.1.1/.built] Error 2
make[3]: Leaving directory '/home/bl/netgear/lede/package/devel/gdb'
time: package/devel/gdb/compile#3.41#1.17#4.41
package/Makefile:107: recipe for target 'package/devel/gdb/compile' failed
make[2]: *** [package/devel/gdb/compile] Error 2
make[2]: Leaving directory '/home/bl/netgear/lede'
package/Makefile:103: recipe for target '/home/bl/netgear/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/bl/netgear/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bl/netgear/lede'
/home/bl/netgear/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

closed 2018-10-06 by coolsnowwolf