coolsnowwolf/lede #554: 编译7620固件时提示menuconfig statement without prompt

tuanqing, 2018-09-22:
make[5]: Entering directory '/home/tuanqing/projects/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.14.71'
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
net/sched/Kconfig:44: warning: menuconfig statement without prompt
*
* Restart config...
*
*
* Composite MTD device drivers
*
Virtual concat MTD device (MTD_VIRT_CONCAT) [N/m/y/?] (NEW)

hcym, 2018-09-23:
选择y，有点不方便 j8，j13之类的。必须中断一下

tuanqing, 2018-09-23:
是不是少打了patch？

madcat0513, 2018-09-23:
编译r7800时也跳出这个，打了个？出现这个
Virtual concat MTD device (MTD_VIRT_CONCAT) [N/m/y/?] (NEW) ?

CONFIG_MTD_VIRT_CONCAT:

This driver allows creation of a virtual MTD concat device, which
concatenates multiple underlying MTD devices to a single device.
This is required by some SoC boards where multiple memory banks are
used as one device with partitions spanning across device boundaries.

Symbol: MTD_VIRT_CONCAT [=n]
Type  : tristate
Prompt: Virtual concat MTD device
  Location:
    -> Device Drivers
      -> Memory Technology Device (MTD) support (MTD [=y])
        -> Composite MTD device drivers
  Defined at drivers/mtd/composite/Kconfig:4
  Depends on: MTD [=y]
应该选y吧

tuanqing, 2018-09-23:
大雕 在今天的源码更新中已解决此问题。

closed 2018-09-23 by tuanqing