coolsnowwolf/lede #559: PHICOMM_K3，R8.1.2 版，Full Cone NAT 错误导致的？

huahuaxue, 2018-09-24:
斐迅K3，无法上网，ttl信息如下:
```
[   13.396146] xt_time: kernel timezone is -0000
[   13.549382] PPP generic driver version 2.4.2
[   13.554748] PPP MPPE Compression module registered
[   13.560717] NET: Registered protocol family 24
[   13.569767] usbcore: registered new interface driver brcmfmac
[   13.575712] pci_generic_config_write32: 48 callbacks suppressed
[   13.575722] pci_bus 0000:01: 1-byte config write to 0000:01:00.0 offset 0x3c may corrupt adjacent RW1C bits
[   13.591459] pci 0000:00:00.0: enabling device (0140 -> 0142)
[   13.597132] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x4 may corrupt adjacent RW1C bits
[   13.606822] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x4 may corrupt adjacent RW1C bits
[   13.616505] brcmfmac 0000:01:00.0: enabling device (0140 -> 0142)
[   13.622620] pci_bus 0000:01: 2-byte config write to 0000:01:00.0 offset 0x4 may corrupt adjacent RW1C bits
[   13.632305] pci_bus 0000:01: 2-byte config write to 0000:01:00.0 offset 0x4 may corrupt adjacent RW1C bits
[   13.810135] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4366c-pcie for chip BCM4366/4
[   13.823608] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac4366c-pcie.txt failed with error -2
[   14.213006] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4366c-pcie for chip BCM4366/4
[   14.221759] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac4366c-pcie.clm_blob failed with error -2
[   14.232066] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   14.243387] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4366/4 wl0: Aug 23 2016 17:19:51 version 10.10.69.69 (r625687) FWID 01-8438621f
[   14.265335] pci_bus 0001:01: 1-byte config write to 0001:01:00.0 offset 0x3c may corrupt adjacent RW1C bits
[   14.275163] pci 0001:00:00.0: enabling device (0140 -> 0142)
[   14.280888] pci_bus 0001:00: 2-byte config write to 0001:00:00.0 offset 0x4 may corrupt adjacent RW1C bits
[   14.290609] pci_bus 0001:00: 2-byte config write to 0001:00:00.0 offset 0x4 may corrupt adjacent RW1C bits
[   14.300330] brcmfmac 0001:01:00.0: enabling device (0140 -> 0142)
[   14.306451] pci_bus 0001:01: 2-byte config write to 0001:01:00.0 offset 0x4 may corrupt adjacent RW1C bits
[   14.316164] pci_bus 0001:01: 2-byte config write to 0001:01:00.0 offset 0x4 may corrupt adjacent RW1C bits
[   14.490226] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4366c-pcie for chip BCM4366/4
[   14.501527] brcmfmac 0001:01:00.0: Direct firmware load for brcm/brcmfmac4366c-pcie.txt failed with error -2
[   14.892804] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4366c-pcie for chip BCM4366/4
[   14.901519] brcmfmac 0001:01:00.0: Direct firmware load for brcm/brcmfmac4366c-pcie.clm_blob failed with error -2
[   14.911833] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   14.923166] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4366/4 wl0: Aug 23 2016 17:19:51 version 10.10.69.69 (r625687) FWID 01-8438621f
[   14.959020] kmodloader: done loading kernel modules from /etc/modules.d/*
[   17.031054] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   18.722407] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   18.730460] bgmac_bcma bcma0:3 eth0: Link is Up - 1Gbps/Full - flow control off
[   18.738026] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   18.745175] br-lan: port 1(eth0.1) entered blocking state
[   18.750622] br-lan: port 1(eth0.1) entered disabled state
[   18.756309] device eth0.1 entered promiscuous mode
[   18.761130] device eth0 entered promiscuous mode
[   18.767217] br-lan: port 1(eth0.1) entered blocking state
[   18.772717] br-lan: port 1(eth0.1) entered forwarding state
[   18.778503] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   19.163807] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   19.799469] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   19.805813] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   19.815751] br-lan: port 2(wlan1) entered blocking state
[   19.821105] br-lan: port 2(wlan1) entered disabled state
[   19.826759] device wlan1 entered promiscuous mode
[   20.128102] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   20.372571] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   20.379070] br-lan: port 2(wlan1) entered blocking state
[   20.384436] br-lan: port 2(wlan1) entered forwarding state
[   22.689173] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   23.736451] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[  580.659665] device wlan1 left promiscuous mode
[  580.664452] br-lan: port 2(wlan1) entered disabled state
[  580.769654] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[  582.057166] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  582.066232] br-lan: port 2(wlan1) entered blocking state
[  582.071592] br-lan: port 2(wlan1) entered disabled state
[  582.077291] device wlan1 entered promiscuous mode
[  582.535988] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  582.542555] br-lan: port 2(wlan1) entered blocking state
[  582.547920] br-lan: port 2(wlan1) entered forwarding state
root@OpenWrt:/# /etc/init.d/network restart
[ 2042.061534] br-lan: port 2(wlan1) entered disabled state
[ 2042.066990] br-lan: port 1(eth0.1) entered disabled state
[ 2042.081821] device eth0.1 left promiscuous mode
[ 2042.086403] device eth0 left promiscuous mode
[ 2042.091001] br-lan: port 1(eth0.1) entered disabled state
[ 2042.101570] IPv6: ADDRCONF(NETDEV_UP): eth0.1: link is not ready
[ 2042.130278] device wlan1 left promiscuous mode
[ 2042.134818] br-lan: port 2(wlan1) entered disabled state
root@OpenWrt:/# [ 2044.002896] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[ 2044.009118] bgmac_bcma bcma0:3 eth0: Link is Up - 1Gbps/Full - flow control off
[ 2044.014388] br-lan: port 1(eth0.1) entered blocking state
[ 2044.014392] br-lan: port 1(eth0.1) entered disabled state
[ 2044.014583] device eth0.1 entered promiscuous mode
[ 2044.014585] device eth0 entered promiscuous mode
[ 2044.015317] br-lan: port 1(eth0.1) entered blocking state
[ 2044.015322] br-lan: port 1(eth0.1) entered forwarding state
[ 2044.081148] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[ 2044.472607] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[ 2044.681920] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[ 2044.691031] br-lan: port 2(wlan1) entered blocking state
[ 2044.696390] br-lan: port 2(wlan1) entered disabled state
[ 2044.702100] device wlan1 entered promiscuous mode
[ 2045.206276] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[ 2045.212938] br-lan: port 2(wlan1) entered blocking state
[ 2045.218292] br-lan: port 2(wlan1) entered forwarding state
[ 2045.776470] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[ 2046.475964] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
```

SouHong, 2018-09-24:
请问固件有WIFI吗？之前编译的没有WIFI项了。

huahuaxue, 2018-09-24:
> 请问固件有WIFI吗？之前编译的没有WIFI项了。

有的。你没有WIFI，是不是你编译的时候，没有勾选K3WIFI那个包？

SouHong, 2018-09-26:
是没有没有勾选K3WIFI，现在正常使用了。

closed 2018-09-28 by coolsnowwolf