coolsnowwolf/lede #564: R7800运行很慢，过一段时间会访问不了。

hxy8241, 2018-09-25:
可能和下面这段log有关。
[   13.139004] mount_root: loading kmods from internal overlay
[   13.160507] kmodloader: loading kernel modules from //etc/modules-boot.d/*
[   13.162142] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
[   13.645897] print_req_error: I/O error, dev mtdblock0, sector 0
[   13.646480] print_req_error: I/O error, dev mtdblock0, sector 8
[   13.651170] print_req_error: I/O error, dev mtdblock0, sector 16
[   13.657141] print_req_error: I/O error, dev mtdblock0, sector 24
[   13.663652] print_req_error: I/O error, dev mtdblock0, sector 0
[   13.668768] Buffer I/O error on dev mtdblock0, logical block 0, async page read
[   13.675464] print_req_error: I/O error, dev mtdblock1, sector 0
[   13.682197] print_req_error: I/O error, dev mtdblock1, sector 8
[   13.688189] print_req_error: I/O error, dev mtdblock1, sector 16
[   13.694064] print_req_error: I/O error, dev mtdblock1, sector 24
[   13.700333] print_req_error: I/O error, dev mtdblock1, sector 0
[   13.705746] Buffer I/O error on dev mtdblock1, logical block 0, async page read
[   14.652468] UBIFS (ubi0:1): default file-system created
[   14.863310] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 145
[   15.992981] UBIFS (ubi0:1): UBIFS: mounted UBI device 0, volume 1, name "rootfs_data"
[   15.993012] UBIFS (ubi0:1): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   15.999790] UBIFS (ubi0:1): FS size: 86851584 bytes (82 MiB, 684 LEBs), journal size 4317184 bytes (4 MiB, 34 LEBs)
[   16.009776] UBIFS (ubi0:1): reserved for root: 4102214 bytes (4006 KiB)
[   16.020007] UBIFS (ubi0:1): media format: w4/r0 (latest is w5/r0), UUID F5E27DA8-5132-4007-B493-3B43B43C9591, small LPT model
[   16.027856] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
[   16.038233] block: unable to load configuration (fstab: Entry not found)
[   16.044908] block: attempting to load /tmp/ubifs_cfg/etc/config/fstab
[   16.051793] block: unable to load configuration (fstab: Entry not found)
[   16.058205] block: attempting to load /etc/config/fstab
[   16.065988] block: unable to load configuration (fstab: Entry not found)
[   16.069806] block: no usable configuration
[   16.119510] UBIFS (ubi0:1): un-mount UBI device 0
[   16.119592] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" stops
[   16.128471] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 146
[   16.705716] UBIFS (ubi0:1): UBIFS: mounted UBI device 0, volume 1, name "rootfs_data"
[   16.705748] UBIFS (ubi0:1): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   16.712618] UBIFS (ubi0:1): FS size: 86851584 bytes (82 MiB, 684 LEBs), journal size 4317184 bytes (4 MiB, 34 LEBs)
[   16.722502] UBIFS (ubi0:1): reserved for root: 4102214 bytes (4006 KiB)
[   16.732746] UBIFS (ubi0:1): media format: w4/r0 (latest is w5/r0), UUID F5E27DA8-5132-4007-B493-3B43B43C9591, small LPT model
[   16.754840] Buffer I/O error on dev mtdblock0, logical block 0, async page read
[   16.757717] Buffer I/O error on dev mtdblock1, logical block 0, async page read
[   17.872520] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
[   17.872644] block: unable to load configuration (fstab: Entry not found)
[   17.878349] block: attempting to load /tmp/ubifs_cfg/etc/config/fstab
[   17.885464] block: unable to load configuration (fstab: Entry not found)
[   17.891622] block: attempting to load /etc/config/fstab
[   17.898506] block: unable to load configuration (fstab: Entry not found)
[   17.903495] block: no usable configuration

下面是完整的kernel log。openwrt.txt是官方的kernel log。
官方的好像没这问题。
[openwrt.txt](https://github.com/coolsnowwolf/lede/files/2416152/openwrt.txt)
[lean.txt](https://github.com/coolsnowwolf/lede/files/2416154/lean.txt)

hcym, 2018-09-25:
有心人，我的7800和7500v2最近的新编译都是有类似故障，不过没看log，直接刷回老版本完事

hxy8241, 2018-09-26:
我也不清楚，只是和官方版本的log用比较软件比较了下，基本一样，就是多出来了这部分。看起来好像是磁盘读取的问题。只能等大神解决了。

coolsnowwolf, 2018-09-27:
MTD ecc 问题，切换到 4.9 就好了

hcym, 2018-09-27:
换4.9编译又出叉子了，什么dts错误

coolsnowwolf, 2018-09-27:
小问题，dts 问题，修复了

hcym, 2018-09-27:
谢谢大雕，明天试试看看，谢谢，看坛子里面热闹，痛苦不堪撒

hcym, 2018-09-28:
编译顺利，还是会丢失lan，哭晕回七月份版本

closed 2018-09-28 by coolsnowwolf