coolsnowwolf/lede #565: Patch failed!  Please fix

maxxwall, 2018-09-27:
同步最新源码后编译PHICOMM K3 报错  Ubuntu 18.04.1 LTS
Hunk #1 FAILED at 285.
1 out of 1 hunk FAILED -- saving rejects to file drivers/tty/serial/earlycon.c.rej
Patch failed!  Please fix /home/msi/lede/target/linux/bcm53xx/patches-4.14/910-Revert-earlycon-initialise-baud-field-of-earlycon-de.patch!
Makefile:31: recipe for target '/home/msi/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/linux-4.14.72/.prepared_82f976a89b0688feb1999474522e620b' failed
make[4]: *** [/home/msi/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/linux-4.14.72/.prepared_82f976a89b0688feb1999474522e620b] Error 1
make[4]: Leaving directory '/home/msi/lede/target/linux/bcm53xx'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/msi/lede/target/linux'
time: target/linux/compile#7.82#2.90#15.40
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/home/msi/lede'
target/Makefile:18: recipe for target '/home/msi/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_compile' failed
make[1]: *** [/home/msi/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/msi/lede'
/home/msi/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

tossp, 2018-09-27:
查看 f18b0b522c5f8f0f5518cae96b44548cddd5e1a4 貌似已经修复了

tossp, 2018-09-27:
@maxxwall 已经编译通过，关闭了吧

maxxwall, 2018-09-27:
@tossp 

抱歉，刚看到，试了下，可以了

closed 2018-09-27 by maxxwall