coolsnowwolf/lede #574: K2T 4.9 编译发生错误 4.14 太卡了

SkyMonkya, 2018-09-28:
错误日志如下:
`SHELL= flock /work/lede/tmp/.root-copy.flock -c 'cp -fpR /work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/packages/.pkgdir/kernel/. /work/lede/staging_dir/target-mips_24kc_musl/root-ath79/'
touch /work/lede/staging_dir/target-mips_24kc_musl/root-ath79/stamp/.kernel_installed
ERROR: module '/work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-4.9.129/drivers/leds/leds-reset.ko' is missing.
/work/lede/target/linux/ath79/modules.mk:16: recipe for target '/work/lede/bin/targets/ath79/generic/packages/kmod-leds-reset_4.9.129-1_mips_24kc.ipk' failed
make[3]: *** [/work/lede/bin/targets/ath79/generic/packages/kmod-leds-reset_4.9.129-1_mips_24kc.ipk] Error 1
make[3]: Leaving directory '/work/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.27#0.24#2.65
package/Makefile:107: recipe for target 'package/kernel/linux/compile' failed
`

SkyMonkya, 2018-09-28:
去掉kernel modules -> LED -modules -> kmod-leds-reset选项之后编译仍然错误:
`/work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/ath10k-ct-2018-08-22-34954f0f/ath10k-4.16/pci.c:3677:2: error: implicit declaration of function 'pci_release_region'; did you mean 'pci_release_regions'? [-Werror=implicit-function-declaration]
  pci_release_region(pdev, BAR_NUM);
  ^~~~~~~~~~~~~~~~~~
  pci_release_regions
cc1: some warnings being treated as errors
scripts/Makefile.build:296: recipe for target '/work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/ath10k-ct-2018-08-22-34954f0f/ath10k-4.16/pci.o' failed
make[5]: *** [/work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/ath10k-ct-2018-08-22-34954f0f/ath10k-4.16/pci.o] Error 1
Makefile:1508: recipe for target '_module_/work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/ath10k-ct-2018-08-22-34954f0f/ath10k-4.16' failed
make[4]: *** [_module_/work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/ath10k-ct-2018-08-22-34954f0f/ath10k-4.16] Error 2
make[4]: Leaving directory '/work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-4.9.129'
Makefile:96: recipe for target '/work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/ath10k-ct-2018-08-22-34954f0f/.built' failed
make[3]: *** [/work/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/ath10k-ct-2018-08-22-34954f0f/.built] Error 2
make[3]: Leaving directory '/work/lede/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/compile#1.73#0.34#2.84
package/Makefile:107: recipe for target 'package/kernel/ath10k-ct/compile' failed
make[2]: *** [package/kernel/ath10k-ct/compile] Error 2
`

coolsnowwolf, 2018-09-28:
K2T不支持 4.9内核

closed 2018-09-28 by coolsnowwolf