coolsnowwolf/lede #580: K3 wifi (含5G) 下，局域网传输文件非常慢

fy1128, 2018-09-29:
基于 commit "[fix ipq806x kernel 4.9 dts](https://github.com/coolsnowwolf/lede/commit/9d0648e37299d3313d7e96d20d7593d54cacabd6)"  编译。

`.config` 配置如下：

```
CONFIG_TARGET_bcm53xx=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_bcm53xx_DEVICE_phicomm-k3=y
CONFIG_TARGET_DEVICE_bcm53xx_DEVICE_phicomm-k3=y
CONFIG_ARIA2_BITTORRENT=y
CONFIG_ARIA2_NOXML=y
CONFIG_ARIA2_OPENSSL=y
CONFIG_ARIA2_WEBSOCKET=y
CONFIG_BUILD_PATENTED=y
CONFIG_KERNEL_KALLSYMS=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_PACKAGE_V2RAY_INCLUDE_GEOIP=y
CONFIG_PACKAGE_V2RAY_INCLUDE_GEOSITE=y
CONFIG_PACKAGE_V2RAY_INCLUDE_V2CTL=y
CONFIG_PACKAGE_V2RAY_INCLUDE_V2RAY=y
CONFIG_PACKAGE_aria2=y
CONFIG_PACKAGE_attr=y
CONFIG_PACKAGE_avahi-dbus-daemon=y
CONFIG_PACKAGE_bind-host=y
CONFIG_PACKAGE_bind-libs=y
CONFIG_PACKAGE_ca-bundle=y
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_dbus=y
CONFIG_PACKAGE_ddns-scripts_cloudflare.com-v4=y
CONFIG_PACKAGE_frpc=y
CONFIG_PACKAGE_hd-idle=y
CONFIG_PACKAGE_ip-full=y
# CONFIG_PACKAGE_ip-tiny is not set
# CONFIG_PACKAGE_kmod-nft-core is not set
# CONFIG_PACKAGE_kmod-nft-nat is not set
# CONFIG_PACKAGE_kmod-nft-offload is not set
CONFIG_PACKAGE_krb5-libs=y
CONFIG_PACKAGE_libattr=y
CONFIG_PACKAGE_libavahi-client=y
CONFIG_PACKAGE_libavahi-dbus-support=y
CONFIG_PACKAGE_libbz2=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcomerr=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libdaemon=y
CONFIG_PACKAGE_libdbus=y
CONFIG_PACKAGE_libexpat=y
CONFIG_PACKAGE_libffi=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libpopt=y
CONFIG_PACKAGE_libss=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libtirpc=y
CONFIG_PACKAGE_libuci-lua=y
CONFIG_PACKAGE_lua-cjson=y
CONFIG_PACKAGE_luci-app-aria2=y
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-hd-idle=y
CONFIG_PACKAGE_luci-app-kcptun=y
CONFIG_PACKAGE_luci-app-samba4=y
CONFIG_PACKAGE_luci-app-transparent-proxy=y
CONFIG_PACKAGE_luci-app-udp2raw=y
CONFIG_PACKAGE_luci-app-v2ray-pro=y
CONFIG_PACKAGE_luci-app-xunlei=y
CONFIG_PACKAGE_luci-i18n-aria2-zh-cn=y
CONFIG_PACKAGE_luci-i18n-hd-idle-zh-cn=y
CONFIG_PACKAGE_luci-i18n-kcptun-zh-cn=y
CONFIG_PACKAGE_luci-i18n-samba4-zh-cn=y
CONFIG_PACKAGE_luci-i18n-v2ray-pro-zh-cn=y
CONFIG_PACKAGE_python-base=y
CONFIG_PACKAGE_python-light=y
CONFIG_PACKAGE_samba4-libs=y
CONFIG_PACKAGE_samba4-server=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_udp2raw-tunnel=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_v2ray=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_VFS=y
```

刷写固件并重置设备后，仅设置接口 wifi (含5G) 及 pppoe 联网。

**其中5G wifi 设置：**
* 国家：US
* 信道：149

**测速工具：**
* scp
* ftp
* samba4

**结果：**

速度平均值低于 400KiB/秒。
**外网速度正常**

SouHong, 2018-09-29:
我这已经正常了。

closed 2018-10-06 by coolsnowwolf

fy1128, 2018-10-11:
最新 [commit](https://github.com/coolsnowwolf/lede/commit/977f6501d605d0d21a39f35c3afa391398301002) 编译，还是这个问题。

怎么调试？

fy1128, 2018-10-11:
```
CONFIG_TARGET_bcm53xx=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_bcm53xx_DEVICE_phicomm-k3=y
CONFIG_TARGET_DEVICE_bcm53xx_DEVICE_phicomm-k3=y

```