coolsnowwolf/lede #619: odhcp6c odhcp6 默认没有勾选

maxxwall, 2018-10-07:
大雕，odhcp6c odhcp6 这两个为什么默认没有勾选了呢？没有这两个IPv6无法正常运行，老早就这样了，但是不敢问你，还有昨天你删了一堆，好心痛啊，每天都打开你的项目看看更新了什么，修复了什么，有更新就git pull 并make，你这样一删，感觉少了什么

我把删之前的备份了，继续git pull，不过出了点问题，有冲突还是什么，菜鸟一只不懂，但我还是试着排查一下

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ddns-scripts_aliyun:
 * 	wget * 	openssl-util * 
 * opkg_install_cmd: Cannot install package ddns-scripts_aliyun.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/msi/lede'
package/Makefile:104: recipe for target '/home/msi/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/msi/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/msi/lede'
/home/msi/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
msi@PC:~/lede$ make menuconfig
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/feeds/packages/getdns/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/feeds/packages/gnurl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/feeds/packages/php7-pecl-http/Makefile' has a dependency on 'libidn2', which does not exist
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

msi@PC:~/lede$

hcym, 2018-10-07:
下床气，过一阵子就会好了，嘎嘎嘎

SkyMonkya, 2018-10-07:
需要支持自己选就行了 默认不支持IPv6 多看看commit 上面有描述