coolsnowwolf/lede #625: K3最新源码出现了软件包冲突，附上编译配置和编译日志

ZGNCZMY, 2018-10-07:
由00209c9bb891d3a18fea6cab55e490434fe01a1c到最新的，配置文件因为上传格式问题我加了.txt后缀，已经在lede/package目录创建了openwrt-packages/lean文件夹的，麻烦大雕了,谢谢
[build.log](https://github.com/coolsnowwolf/lede/files/2454085/build.log)
[20181008.config.txt](https://github.com/coolsnowwolf/lede/files/2454088/20181008.config.txt)

maxxwall, 2018-10-08:
我这边没问题，
我建议你git pull后删除.config 配置文件并 make clean
再编译

ZGNCZMY, 2018-10-08:
> 
> 
> 我这边没问题，
> 我建议你git pull后删除.config 配置文件并 make clean
> 再编译

谢谢，我这边先自己想想办法

ZGNCZMY, 2018-10-08:
我把大雕私人旧包luci-app-openvpn-server从package/lean/openwrt-packages/复制了一份到package/lean/，这样没报错成功过了

ZGNCZMY, 2018-10-08:
> 
> 
> @ZGNCZMY 这样啊，抱歉，我菜鸟一只，不该乱建议，抱歉

没有啊.我也是小菜鸟

waybq, 2018-10-11:
你好，我编译K3时不成功，提示如下，不知该怎么调。不知你们碰到过没

`
make[5]: Entering directory '/wadata/backup/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/linux-4.14.75'
scripts/kconfig/conf  --silentoldconfig Kconfig
net/sched/Kconfig:44: warning: menuconfig statement without prompt
*
* Restart config...
*
*
* Compile-time checks and compiler options
*
Allow gcc to uninline functions marked 'inline' (OPTIMIZE_INLINING) [N/y/?] (NEW) aborted!

Console input/output is redirected. Run 'make oldconfig' to update configuration.

scripts/kconfig/Makefile:38: recipe for target 'silentoldconfig' failed
make[7]: *** [silentoldconfig] Error 1
Makefile:521: recipe for target 'silentoldconfig' failed
make[6]: *** [silentoldconfig] Error 2

The present kernel configuration has modules disabled.
Type 'make config' and enable loadable module support.
Then build a kernel with module support enabled.

Makefile:1286: recipe for target 'modules' failed
make[5]: *** [modules] Error 1
make[5]: Leaving directory '/wadata/backup/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/linux-4.14.75'
Makefile:31: recipe for target '/wadata/backup/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/linux-4.14.75/.modules' failed
make[4]: *** [/wadata/backup/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx/linux-4.14.75/.modules] Error 2
make[4]: Leaving directory '/wadata/backup/openwrt/lede/target/linux/bcm53xx'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/wadata/backup/openwrt/lede/target/linux'
time: target/linux/compile#2.54#0.64#20.80
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/wadata/backup/openwrt/lede'
target/Makefile:18: recipe for target '/wadata/backup/openwrt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_compile' failed
make[1]: *** [/wadata/backup/openwrt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/wadata/backup/openwrt/lede'
/wadata/backup/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
`

closed 2018-10-12 by coolsnowwolf