coolsnowwolf/lede #627: 更新10月1日版本后路由器不能分配ipv6地址

JosephCN, 2018-10-07:
已经多次确认配置和恢复默认设置，终端均不能获取到ipv6地址，江苏电信能正常下发并获取ipv6pd前缀

SkyMonkya, 2018-10-08:
默认编译版本不支持IPv6 需要支持自己添加支持包

LitCcc, 2018-10-08:
我也发现这个问题了，就算安装opdhcp6之后仍然不会分配给终端

SkyMonkya, 2018-10-08:
需要两个包吧 没记错的话

LitCcc, 2018-10-08:
> 需要两个包吧 没记错的话

这样啊，我回去看看

LitCcc, 2018-10-08:
是的，需要odhcp6和odhcpd两个包即可解决

获取 Outlook for iOS<https://aka.ms/o0ukef>

________________________________
收件人: SkyMonkya <notifications@github.com>
发送时间: 星期一, 十月 8, 2018 11:58
收件人: coolsnowwolf/lede
抄送: LitCcc; Comment
主题: Re: [coolsnowwolf/lede] 更新10月1日版本后路由器不能分配ipv6地址 (#627)

需要两个包吧 没记错的话

madcat0513, 2018-10-09:
安装odhcp6c并且dnsmasq里打开dhcpv6支持就能正常获得ipv6了

JosephCN, 2018-10-10:
感谢各位确认odhcp6和odhcpd这两个包正确安装即可解决问题。

closed 2018-10-10 by JosephCN

zcluo, 2018-10-10:
安装后要设置什么嘛在dhcp那个页面？

发自我的 iPhone

madcat0513, 2018-10-10:
> 感谢各位确认odhcp6和odhcpd这两个包正确安装即可解决问题。

 x x                                                                              <*> odhcp6c. Embedded DHCPv6-client for OpenWrt                                                                                                     x x
  x x                                                                              (0)   CER-ID Extension ID (0 = disabled)                                                                                                            x x
  x x                                                                              < > odhcpd. OpenWrt DHCPv6(-PD)/RA Server & Relay and DHCPv4 server                                                                                 x x
  x x                                                                              < > odhcpd-ipv6only. OpenWrt DHCPv6(-PD)/RA Server & Relay                                                                                          x x
就看见这三个包，勾哪两个？