coolsnowwolf/lede #633: R7800添加访客网络后访客网络无法联网且luci界面上显示异常

qyh214, 2018-10-08:
主机型号 | Netgear Nighthawk X4S R7800
-- | --
固件版本 | OpenWrt  R8.1.6 By Lean / 			LuCI openwrt-18.06 branch (git-18.138.53009-80cb4fe)
内核版本 | 4.9.131

编译使用的commit：b05f470

开启了Turbo ACC的dns加速，启用 SFE Fast Path 加速

根据openwrt教程在luci界面里添加了访客网络，访客网络无法联网。路由器重启后，均不能联网了并且luci界面奇卡无比，顶着卡顿停止了加速，症状缓解，但是发现首页显示的概况里的ipv4 wan状态被覆盖成了访客网络的信息（我自己上网的是192.168.2.1 但是访客网络是192.168.3.1，但此时发现显示的是192.168.3.1而不是192.168.2.1）

现在只能先删除了访客网络，并正常开启了Turbo ACC的dns加速，启用 SFE Fast Path 加速，使用没有问题，luci也没有出现界面卡顿。

madcat0513, 2018-10-09:
看看wireless的访客网络配置里面是否少了 option network 'lan'这句

madcat0513, 2018-10-11:
按照这个教程搞好了访客网络，加速全开，一切正常啊

qyh214, 2018-10-16:
> 
> 
> 按照这个教程搞好了访客网络，加速全开，一切正常啊

谢谢，不过刷了最新的 0ff0ac5cef90164480bbeaa0caa92f36b056a7c7 已经可以正确使用，没有问题

closed 2018-10-16 by qyh214