coolsnowwolf/lede #635: 勾选了几个luci，编译不了

maxxwall, 2018-10-08:
主机型号 | PHICOMM K3
------------ | -------------

看着是不是缺少文件
[201810082100build.txt](https://github.com/coolsnowwolf/lede/files/2455994/201810082100build.txt)
[k3.config.txt](https://github.com/coolsnowwolf/lede/files/2455995/k3.config.txt)
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for doxygen... no
checking pkg-config is at least version 0.9.0... yes
checking for protobuf >= 2.6.0... no
configure: error: Package requirements (protobuf >= 2.6.0) were not met:

No package 'protobuf' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables protobuf_CFLAGS
and protobuf_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
Makefile:67: recipe for target '/home/msi/lede/build_dir/hostpkg/libprotobuf-c-v1.2.1/.configured' failed
make[3]: *** [/home/msi/lede/build_dir/hostpkg/libprotobuf-c-v1.2.1/.configured] Error 1
make[3]: Leaving directory '/home/msi/lede/feeds/packages/libs/protobuf-c'
time: package/feeds/packages/protobuf-c/host-compile#9.43#0.66#12.04
package/Makefile:107: recipe for target 'package/feeds/packages/protobuf-c/host/compile' failed
make[2]: *** [package/feeds/packages/protobuf-c/host/compile] Error 2
make[2]: Leaving directory '/home/msi/lede'
package/Makefile:103: recipe for target '/home/msi/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/msi/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/msi/lede'
/home/msi/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

huasuifengqu, 2018-10-08:
貌似是你自己的linux系统里少了protobuf这个包

maxxwall, 2018-10-09:
@huasuifengqu 老哥经你提醒，我装了3.0，又按教程装了3.6.1，问题依旧，算了太笨，还是把原来钩的取消算了，弄不了

huasuifengqu, 2018-10-09:
@maxxwall 我查看了一下我系统里安装的包，安装的是protobuf-c这个包