coolsnowwolf/lede #638: samba访问usb设备无法显示中文文件和文件夹

pk49800, 2018-10-09:
**问题描述：使用PC和手机通过samba协议无法显示中文文件夹和文件，此前使用（华硕）官方固件可正常显示。
设备：AC-58U
挂载设备：NTFS分区格式USB3.0移动硬盘
测试设备：WIN10家庭版，iphoneSE IOS12
samba版本：3.6.25-10**

**挂载点如下图**
[image]

**samba共享目录设置如下图（已注释掉invalid users = root，并后台添加root访问账户）**
[image]

**PC通过samba协议访问usb设备显示情况如下图所示：**
sda1根目录所有中文文件夹与文件全部无法显示
[image]
sda2 Game目录文件全英文可正常显示
[image]
[image]
sda2 Movie目录全中文文件无法显示
[image]

**系统日志如下：**
Tue Oct  9 08:43:02 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 94:65:2d:8f:f3:57
Tue Oct  9 08:43:02 2018 daemon.info hostapd: wlan1: STA 94:65:2d:8f:f3:57 WPA: pairwise key handshake completed (RSN)
Tue Oct  9 08:43:02 2018 daemon.info dnsmasq-dhcp[2661]: DHCPDISCOVER(br-lan) 94:65:2d:8f:f3:57
Tue Oct  9 08:43:02 2018 daemon.info dnsmasq-dhcp[2661]: DHCPOFFER(br-lan) 192.168.1.233 94:65:2d:8f:f3:57
Tue Oct  9 08:43:02 2018 daemon.info dnsmasq-dhcp[2661]: DHCPREQUEST(br-lan) 192.168.1.233 94:65:2d:8f:f3:57
Tue Oct  9 08:43:02 2018 daemon.info dnsmasq-dhcp[2661]: DHCPACK(br-lan) 192.168.1.233 94:65:2d:8f:f3:57 OnePlus_5
Tue Oct  9 08:43:53 2018 daemon.info hostapd: wlan1: STA 94:65:2d:8f:f3:57 IEEE 802.11: disconnected due to excessive missing ACKs
Tue Oct  9 08:43:53 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 94:65:2d:8f:f3:57
Tue Oct  9 08:44:13 2018 daemon.info hostapd: wlan1: STA 94:65:2d:8f:f3:57 IEEE 802.11: authenticated
Tue Oct  9 08:44:13 2018 daemon.info hostapd: wlan1: STA 94:65:2d:8f:f3:57 IEEE 802.11: associated (aid 1)
Tue Oct  9 08:44:13 2018 kern.warn kernel: [18732.093260] ath10k_ahb a800000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
Tue Oct  9 08:44:13 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 94:65:2d:8f:f3:57
Tue Oct  9 08:44:13 2018 daemon.info hostapd: wlan1: STA 94:65:2d:8f:f3:57 WPA: pairwise key handshake completed (RSN)
Tue Oct  9 08:44:13 2018 daemon.info dnsmasq-dhcp[2661]: DHCPDISCOVER(br-lan) 94:65:2d:8f:f3:57
Tue Oct  9 08:44:13 2018 daemon.info dnsmasq-dhcp[2661]: DHCPOFFER(br-lan) 192.168.1.233 94:65:2d:8f:f3:57
Tue Oct  9 08:44:13 2018 daemon.info dnsmasq-dhcp[2661]: DHCPREQUEST(br-lan) 192.168.1.233 94:65:2d:8f:f3:57
Tue Oct  9 08:44:13 2018 daemon.info dnsmasq-dhcp[2661]: DHCPACK(br-lan) 192.168.1.233 94:65:2d:8f:f3:57 OnePlus_5
Tue Oct  9 08:46:59 2018 daemon.info hostapd: wlan1: STA 94:65:2d:8f:f3:57 IEEE 802.11: disconnected due to excessive missing ACKs
Tue Oct  9 08:46:59 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 94:65:2d:8f:f3:57
Tue Oct  9 08:47:29 2018 daemon.info hostapd: wlan1: STA 94:65:2d:8f:f3:57 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Oct  9 08:57:34 2018 kern.warn kernel: [19532.971493] ath10k_ahb a800000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
Tue Oct  9 08:57:34 2018 daemon.info hostapd: wlan1: STA 48:a9:d2:a9:ee:14 IEEE 802.11: authenticated
Tue Oct  9 08:57:34 2018 daemon.info hostapd: wlan1: STA 48:a9:d2:a9:ee:14 IEEE 802.11: associated (aid 1)
Tue Oct  9 08:57:34 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 48:a9:d2:a9:ee:14
Tue Oct  9 08:57:34 2018 daemon.info hostapd: wlan1: STA 48:a9:d2:a9:ee:14 WPA: pairwise key handshake completed (RSN)
Tue Oct  9 08:57:35 2018 daemon.info dnsmasq-dhcp[2661]: DHCPREQUEST(br-lan) 192.168.1.146 48:a9:d2:a9:ee:14
Tue Oct  9 08:57:35 2018 daemon.info dnsmasq-dhcp[2661]: DHCPACK(br-lan) 192.168.1.146 48:a9:d2:a9:ee:14 WhaleyTV-bbc8bca55c73c158
Tue Oct  9 09:00:00 2018 cron.info crond[1163]: USER root pid 6526 cmd /usr/share/dnscache-watchdog.sh
Tue Oct  9 09:09:15 2018 daemon.info hostapd: wlan1: STA 74:c1:4f:a7:2a:ef IEEE 802.11: authenticated
Tue Oct  9 09:09:15 2018 daemon.info hostapd: wlan1: STA 74:c1:4f:a7:2a:ef IEEE 802.11: associated (aid 3)
Tue Oct  9 09:09:15 2018 kern.warn kernel: [20234.474222] ath10k_ahb a800000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
Tue Oct  9 09:09:15 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 74:c1:4f:a7:2a:ef
Tue Oct  9 09:09:15 2018 daemon.info hostapd: wlan1: STA 74:c1:4f:a7:2a:ef WPA: pairwise key handshake completed (RSN)
Tue Oct  9 09:09:16 2018 daemon.info dnsmasq-dhcp[2661]: DHCPREQUEST(br-lan) 192.168.1.176 74:c1:4f:a7:2a:ef
Tue Oct  9 09:09:16 2018 daemon.info dnsmasq-dhcp[2661]: DHCPACK(br-lan) 192.168.1.176 74:c1:4f:a7:2a:ef
Tue Oct  9 09:09:16 2018 daemon.info dnsmasq-dhcp[2661]: DHCPREQUEST(br-lan) 192.168.1.176 74:c1:4f:a7:2a:ef
Tue Oct  9 09:09:16 2018 daemon.info dnsmasq-dhcp[2661]: DHCPACK(br-lan) 192.168.1.176 74:c1:4f:a7:2a:ef
Tue Oct  9 09:28:20 2018 daemon.info hostapd: wlan1: STA 98:22:ef:ca:bc:b3 IEEE 802.11: authenticated
Tue Oct  9 09:28:20 2018 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 98:22:ef:ca:bc:b3 1
Tue Oct  9 09:28:20 2018 daemon.info hostapd: wlan1: STA 98:22:ef:ca:bc:b3 IEEE 802.11: associated (aid 4)
Tue Oct  9 09:28:20 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 98:22:ef:ca:bc:b3
Tue Oct  9 09:28:20 2018 daemon.info hostapd: wlan1: STA 98:22:ef:ca:bc:b3 WPA: pairwise key handshake completed (RSN)
Tue Oct  9 09:28:20 2018 kern.warn kernel: [21379.135753] ath10k_ahb a800000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
Tue Oct  9 09:28:20 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 98:22:ef:ca:bc:b3
Tue Oct  9 09:28:20 2018 daemon.info hostapd: wlan1: STA 98:22:ef:ca:bc:b3 IEEE 802.11: disassociated
Tue Oct  9 09:28:21 2018 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 98:22:ef:ca:bc:b3 1
Tue Oct  9 09:28:21 2018 daemon.info hostapd: wlan1: STA 98:22:ef:ca:bc:b3 IEEE 802.11: authenticated
Tue Oct  9 09:28:21 2018 daemon.info hostapd: wlan1: STA 98:22:ef:ca:bc:b3 IEEE 802.11: associated (aid 4)
[... 1199 lines cut ...]
[29760.955480] ntfs: (device sda2): ntfs_ucstonls(): Unicode name contains characters that cannot be converted to character set iso8859-1.  You might want to try to use the mount option nls=utf8.
[29760.955551] ntfs: (device sda2): ntfs_filldir(): Skipping unrepresentable inode 0xeaf6.
[29760.972015] ntfs: (device sda2): ntfs_ucstonls(): Unicode name contains characters that cannot be converted to character set iso8859-1.  You might want to try to use the mount option nls=utf8.
[29760.979535] ntfs: (device sda2): ntfs_filldir(): Skipping unrepresentable inode 0xeb02.
[29760.996922] ntfs: (device sda2): ntfs_ucstonls(): Unicode name contains characters that cannot be converted to character set iso8859-1.  You might want to try to use the mount option nls=utf8.
[29761.004640] ntfs: (device sda2): ntfs_filldir(): Skipping unrepresentable inode 0x77ae.
[29761.021914] ntfs: (device sda2): ntfs_ucstonls(): Unicode name contains characters that cannot be converted to character set iso8859-1.  You might want to try to use the mount option nls=utf8.
[29761.029636] ntfs: (device sda2): ntfs_filldir(): Skipping unrepresentable inode 0x48.
[29832.415846] __ntfs_error: 8 callbacks suppressed
[29832.415988] ntfs: (device sda2): ntfs_ucstonls(): Unicode name contains characters that cannot be converted to character set iso8859-1.  You might want to try to use the mount option nls=utf8.
[29832.420010] ntfs: (device sda2): ntfs_filldir(): Skipping unrepresentable inode 0xeaf6.
[29832.436991] ntfs: (device sda2): ntfs_ucstonls(): Unicode name contains characters that cannot be converted to character set iso8859-1.  You might want to try to use the mount option nls=utf8.
[29832.444688] ntfs: (device sda2): ntfs_filldir(): Skipping unrepresentable inode 0xeb02.
[29832.462023] ntfs: (device sda2): ntfs_ucstonls(): Unicode name contains characters that cannot be converted to character set iso8859-1.  You might want to try to use the mount option nls=utf8.
[29832.469727] ntfs: (device sda2): ntfs_filldir(): Skipping unrepresentable inode 0x77ae.
[29832.487108] ntfs: (device sda2): ntfs_ucstonls(): Unicode name contains characters that cannot be converted to character set iso8859-1.  You might want to try to use the mount option nls=utf8.
[29832.494836] ntfs: (device sda2): ntfs_filldir(): Skipping unrepresentable inode 0x48.
[30462.085355] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
[30462.086281] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
[30462.093532] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
[30462.101330] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
[30462.107353] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
[30462.114624] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
[30462.121866] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
[30462.128837] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon

pk49800, 2018-10-09:
已解决，固件编译时添加对应的语言支持即可

closed 2018-10-09 by pk49800

f3404923, 2021-07-01:
兄弟是在Kernel module->Native language support里面的吗？
[image]