coolsnowwolf/lede #641: 编译k2固件出一个错误，解决了

tianlichunhong, 2018-10-09:
错误是you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this，
解决办法是输入一条命令export FORCE_UNSAFE_CONFIGURE=1，解决了。

tianlichunhong, 2018-10-10:
k2的编译到最后提示
WARNING: Image file /home/labasky/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-ex2700-squashfs-sysupgrade.bin is too big
在官网查到需要用下面命令解决
make -j1 V=s TARGET_DEVICES=y1s

tianlichunhong, 2018-10-10:
还有提示fatal: The remote end hung up unexpectedly
通过在lede目录输入git config http.postBuffer 524288000

waybq, 2018-10-10:
你好，我编译也没报错，结束后在目录中lede/bin/targets/ramips/mt7620中找不到K2的固件，不知这是什么原因。

```
Generating index for package ./wget_1.19.5-1_mipsel_24kc.ipk
Generating index for package ./wifischedule_1-2_all.ipk
Signing package index...
make[2]: Leaving directory '/openwrt/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/openwrt/lede'
make[2]: Leaving directory '/openwrt/lede'
make[1]: Leaving directory '/openwrt/lede'

```

tianlichunhong, 2018-10-11:
> 你好，我编译也没报错，结束后在目录中lede/bin/targets/ramips/mt7620中找不到K2的固件，不知这是什么原因。
>
> ```
> Generating index for package ./wget_1.19.5-1_mipsel_24kc.ipk
> Generating index for package ./wifischedule_1-2_all.ipk
> Signing [...]

在target images选择psg1218，我之前没选。

Joelyi, 2018-10-14:
不知你编译是选的4.14内核吗？我编译选了4.9内核之后，breed刷进路由无限自动重启，4.14内核就没问题

tianlichunhong, 2018-10-15:
> 不知你编译是选的4.14内核吗？我编译选了4.9内核之后，breed刷进路由无限自动重启，4.14内核就没问题

我是默认，没有更改其他选项

closed 2018-11-27 by coolsnowwolf

hnlyxjb, 2019-11-26:
> k2的编译到最后提示
> WARNING: Image file /home/labasky/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-ex2700-squashfs-sysupgrade.bin is too big
> 
> 原来是target Images 没有选择。选择PSG1218就好。
你好，我和你出现的错误一样但是你说的target Image里面只有几个简单的选项
ramdisk
 cpio.gz 
tar.gz
 ext4
 squashfs
没有别的选项了

ZqinKing, 2020-01-08:
to big，选的插件太多了，编译出来的固件超过8M了