coolsnowwolf/lede #662: 编译有异常

Timerun16ning, 2018-10-12:
默认编译k2固件
报错：
tar: linux-4.14.75/virt/lib/irqbypass.c: Cannot open: No such file or directory
tar: Exiting with failure status due to previous errors
Makefile:104: recipe for target '/home/ubuntu/lede/build_dir/toolchain-mipsel_24kc_gcc-7.3.0_musl/linux-4.14.75/.prepared' failed
make[3]: *** [/home/ubuntu/lede/build_dir/toolchain-mipsel_24kc_gcc-7.3.0_musl/linux-4.14.75/.prepared] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#28.09#7.68#138.07
toolchain/Makefile:96: recipe for target 'toolchain/kernel-headers/compile' failed
make[2]: *** [toolchain/kernel-headers/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
toolchain/Makefile:94: recipe for target '/home/ubuntu/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.3.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
ubuntu@ip-172-31-47-216:~/lede$ 
请教作者怎么解决，谢谢你的开源代码

coolsnowwolf, 2018-10-12:
不要用 root 用户 git clone 和 编译

Timerun16ning, 2018-10-14:
没有使用root用户编译

Timerun16ning, 2018-10-20:
挂全局梯子编译

closed 2018-10-20 by Timerun16ning