coolsnowwolf/lede #677: AC58U升级8.1.7以后5G信号异常

pk49800, 2018-10-15:
AC58U升级8.1.7以后5G信号异常，只有贴近路由器才能搜到5G，距离拉远到2米以后连接5G提示密码不正确
[image]
[image]

pk49800, 2018-10-15:
**以下是系统日志：**
Mon Oct 15 01:34:08 2018 daemon.err dnsmasq[6912]: failed to load names from /tmp/hosts/dhcp.cfg01411c: Permission denied
Mon Oct 15 01:34:08 2018 daemon.info dnsmasq-dhcp[6912]: read /etc/ethers - 0 addresses
Mon Oct 15 01:54:41 2018 kern.warn kernel: [ 1363.241637] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.238809] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.239477] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.245410] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.253053] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.260243] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.267634] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.274579] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.282101] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.289363] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 01:55:16 2018 kern.warn kernel: [ 1399.296720] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
Mon Oct 15 02:00:00 2018 cron.info crond[6284]: USER root pid 7257 cmd /usr/share/dnscache-watchdog.sh
Mon Oct 15 02:03:37 2018 daemon.warn dnsmasq[6912]: possible DNS-rebind attack detected: preview.product.10046.mi.com
Mon Oct 15 02:03:37 2018 daemon.warn dnsmasq[6912]: possible DNS-rebind attack detected: preview.product.10046.mi.com
Mon Oct 15 03:00:00 2018 cron.info crond[6284]: USER root pid 7585 cmd /usr/share/dnscache-watchdog.sh
Mon Oct 15 04:00:00 2018 cron.info crond[6284]: USER root pid 7774 cmd /usr/share/dnscache-watchdog.sh
Mon Oct 15 04:11:34 2018 daemon.info dnsmasq-dhcp[6912]: DHCPREQUEST(br-lan) 192.168.1.197 0c:d8:6c:99:d9:63
Mon Oct 15 04:11:34 2018 daemon.info dnsmasq-dhcp[6912]: DHCPACK(br-lan) 192.168.1.197 0c:d8:6c:99:d9:63 FW313R
Mon Oct 15 04:28:31 2018 daemon.info dnsmasq-dhcp[6912]: DHCPREQUEST(br-lan) 192.168.1.149 f4:83:cd:c8:8a:93
Mon Oct 15 04:28:31 2018 daemon.info dnsmasq-dhcp[6912]: DHCPACK(br-lan) 192.168.1.149 f4:83:cd:c8:8a:93 TL-WR842N
Mon Oct 15 05:00:00 2018 cron.info crond[6284]: USER root pid 7967 cmd /usr/share/dnscache-watchdog.sh
Mon Oct 15 06:00:00 2018 cron.info crond[6284]: USER root pid 8113 cmd /usr/share/dnscache-watchdog.sh
Mon Oct 15 06:00:00 2018 cron.info crond[6284]: USER root pid 8114 cmd /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1
Mon Oct 15 06:00:31 2018 cron.info crond[8179]: crond (busybox 1.29.2) started, log level 5
Mon Oct 15 06:00:31 2018 daemon.info dnsmasq[6912]: exiting on receipt of SIGTERM
Mon Oct 15 06:00:31 2018 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Oct 15 06:00:31 2018 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: started, version 2.80test7 cachesize 1024
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: DNS service limited to local subnets
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-auth no-DNSSEC loop-detect inotify dumpfile
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq-dhcp[8268]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain test
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain onion
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain localhost
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain local
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain invalid
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain bind
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using nameserver 127.0.0.1#5333
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain lan
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: reading /tmp/resolv.conf.auto
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain test
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain onion
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain localhost
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain local
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain invalid
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain bind
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using nameserver 127.0.0.1#5333
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using local addresses only for domain lan
Mon Oct 15 06:00:36 2018 daemon.info dnsmasq[8268]: using nameserver 202.96.134.33#53
[... cut ...]
[33643.033389] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 1
[33645.422863] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[33645.427804] br-lan: port 3(wlan1) entered blocking state
[33645.427863] br-lan: port 3(wlan1) entered disabled state
[33645.433900] device wlan1 entered promiscuous mode
[33646.161419] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[33646.161916] br-lan: port 3(wlan1) entered blocking state
[33646.166890] br-lan: port 3(wlan1) entered forwarding state

hcym, 2018-10-15:
以前是有这个情况，5G信号非常弱。刚刚编译是正常的

closed 2018-11-27 by coolsnowwolf