coolsnowwolf/lede #681: SSR plus服务器连接OK但是不能fq

LoogCP, 2018-10-16:
在状态里面谷歌连通性检查显示连接错误,服务器端口检查显示OK,不能翻墙
ssh用/etc/init.d/shadowsocksr restart输出
```
root@OpenWrt:~# /etc/init.d/shadowsocksr restart  
udhcpc: started, v1.29.2  
udhcpc: sending discover
udhcpc: no lease, failing
sh: 1: unknown operand
ipset v6.38: Element cannot be added to the set: it's already added
ipset v6.38: Element cannot be added to the set: it's already added
ipset v6.38: Element cannot be added to the set: it's already added
ipset v6.38: Element cannot be added to the set: it's already added
ipset v6.38: Element cannot be added to the set: it's already added
sh: 1: unknown operand
sh: 1: unknown operand
 2018-10-16 12:53:36 INFO: protocol auth_chain_b
 2018-10-16 12:53:36 INFO: protocol_param 
 2018-10-16 12:53:36 INFO: method aes-256-ctr
 2018-10-16 12:53:36 INFO: obfs tls1.2_ticket_auth
 2018-10-16 12:53:36 INFO: obfs_param 
udhcpc: started, v1.29.2
udhcpc: sending discover
udhcpc: no lease, failing
root@OpenWrt:~# 
```

服务器是OK的,手机上能连能用,固件版本是OpenWrt 18.06.1, r827-5f9650a5,R8.1.7

coolsnowwolf, 2018-10-16:
重启即可，估计你是升级过来的，把老版本配置带过了

LoogCP, 2018-10-16:
> 重启即可，估计你是升级过来的，把老版本配置带过了

我升级的时候是选择了不保留配置,后面所有配置都是自己改的,中间重启过,还是不行

LoogCP, 2018-10-16:
> 重启即可，估计你是升级过来的，把老版本配置带过了

老版本的配置我只把DHCP和network的用SCP传到路由器里边,其他配置都没用老版本的

nimasi, 2018-10-17:
订阅功能可以正常使用吗

LoogCP, 2018-10-17:
> 订阅功能可以正常使用吗

订阅能用，但是订阅服务列表里面的服务器部分OK，部分error，但是不管选哪个服务器，都过不了连通性测试，访问不了Google之类的网站

closed 2018-11-27 by coolsnowwolf

chinawenxu, 2020-03-11:
一样的问题。服务器是好的，但不能翻墙。

self7414386, 2020-03-17:
ros 加openwrt ，ssr plus显示运行，状态中显示可以连接谷歌，却不能连接是什么原因？

dpwfor, 2020-03-21:
一样的问题，服务器正常，手机测试可以，路由器设置，连通性失败，怎么回事

coofly, 2020-03-23:
我也遇到了这样的问题，这是什么情况？

dpwfor, 2020-03-30:
节点的问题，换了节点解决的

发自我的iPhone

wo673060519, 2020-04-01:
解决了吗，咋解决的

dpwfor, 2020-04-01:
节点问题，换了我的节点可以了

发自我的iPhone

wo673060519, 2020-04-01:
> 节点问题，换了我的节点可以了 发自我的iPhone
> […](#)

我的是VPS啊，没法换啊。应该不是节点的问题吧

dpwfor, 2020-04-01:
我的是换节点解决的，你不行可以用我机场的节点测试

发自我的iPhone

wo673060519, 2020-04-01:
> 节点问题，换了我的节点可以了 发自我的iPhone
> […](#)

而且我的昨天还能正常用的，今天没啥征兆就不行了，但是手机连接可以用，公司电脑也能用，我怀疑应该是openwrt的问题。

dpwfor, 2020-04-01:
那可能就是固件的问题，建议换版本

发自我的iPhone

wo673060519, 2020-04-01:
> 那可能就是固件的问题，建议换版本 发自我的iPhone
> […](#)

嗯，多谢兄弟解答。

SundarZhou, 2020-04-30:
有大佬解决这个问题解决了吗，我也是加了正常能使用的服务器，但是测试连接失败

simpleyang1, 2022-05-29:
同样的问题，订阅也可以正常更新，测试也是好的，但是就是不能翻墙，不知道哪里问题，新刷了系统进r2s然后更新的订阅。固件是这个

klever1988/nanopi-openwrt 2022-05-26
--

[klever1988/nanopi-openwrt](https://github.com/klever1988/nanopi-openwrt) 2022-05-26

tesssar, 2022-06-30:
> 同样的问题，订阅也可以正常更新，测试也是好的，但是就是不能翻墙，不知道哪里问题，新刷了系统进r2s然后更新的订阅。固件是这个
> 
> ## klever1988/nanopi-openwrt 2022-05-26
> [klever1988/nanopi-openwrt](https://github.com/klever1988/nanopi-openwrt) 2022-05-26

哈哈，我也是同样的问题，Google了一下跳转进来发现你用的也是1988的固件。我刷了两个版本的固件了都有同样的问题，节点是没问题的，因为同样的节点改用vssr就没问题了。

hunter8605, 2022-09-10:
我也是同样的问题，头大头大

tesssar, 2022-09-10:
用了几个月vssr感觉也不错，跟ssrp差不多

tesssar, 2022-09-10:
> 我也是同样的问题，头大头大

用了几个月vssr感觉也不错，跟ssrp差不多

hunter8605, 2022-09-10:
请问支持hy协议吗？

RaphaelDante, 2023-10-11:
我也遇到这个问题了，跟你一样的情况，而且还另外试了passwall和clash，都不行，之前用passwall一直没问题的，请问怎么解决的

zeroworld1, 2024-05-08:
一样的问题，trojan协议，ssr plus可以连上服务器，并且端口也都测试ok（百度和google），但是无法翻墙。
用passwall和openclash也是一样无法翻墙。怀疑是墙高了的原因。

Jogmann, 2024-11-09:
清除一下DNS缓存：调出CMD，  ipconfig/flushDNS

qian-cr, 2025-01-05:
> 一样的问题，trojan协议，ssr plus可以连上服务器，并且端口也都测试ok（百度和google），但是无法翻墙。 用passwall和openclash也是一样无法翻墙。怀疑是墙高了的原因。

解决了吗？我也一样，可以运行，百度和Google测试也OK，但就是翻不出去

qian-cr, 2025-01-05:
我也是这样，但是发现其他外网都能连，唯独Google、YouTube连不上

qian-cr, 2025-01-05:
难道墙又高了？除了Google、主流媒体不能访问，其他外网可以

ZerOri, 2025-02-28:
已经放弃路由器插件翻墙了