coolsnowwolf/lede #686: r7800无线驱动的选择

madcat0513, 2018-10-16:
< > ath10k-firmware-qca9984. ath10k firmware for QCA9984 devices                                                                                    
                                                                                < > ath10k-firmware-qca9984. ath10k qca9984 firmware                                                                                                
                                                                               <*> ath10k-firmware-qca9984-ct... ath10k CT 10.4 firmware for QCA9984 devices                                                                       
                                                                                < > ath10k-firmware-qca9984-ct-htt..... ath10k CT 10.4 htt-mgt fw for QCA9984   
现在R7800的无线驱动选哪个啊？貌似第一第二两个是一样的，重复了？

coolsnowwolf, 2018-10-16:
rm -rf ./tmp

madcat0513, 2018-10-17:
重复的情况果然搞定，感谢lean，话说现在ct的驱动还是不怎么好用吧，半年前试过，一直不稳定

closed 2018-11-27 by coolsnowwolf