coolsnowwolf/lede #688: 添加set wireless.radio${devidx}.noscan=1后AR71XX编译后WiFi被关闭

Lancenas, 2018-10-17:
uci -q batch <<-EOF
                        set wireless.radio${devidx}=wifi-devicei
                        set wireless.radio${devidx}.type=mac80211
                        set wireless.radio${devidx}.channel=${channel}
                        set wireless.radio${devidx}.hwmode=11${mode_band}
                        ${dev_id}
                        ${ht_capab}
                        set wireless.radio${devidx}.disabled=0
                         uci -q batch <<-EOF
                        set wireless.radio${devidx}=wifi-devicei
                        set wireless.radio${devidx}.type=mac80211
                        set wireless.radio${devidx}.channel=${channel}
                        set wireless.radio${devidx}.hwmode=11${mode_band}
                        ${dev_id}
                        ${ht_capab}
                        set wireless.radio${devidx}.disabled=0
                        set wireless.radio${devidx}.noscan=1

Lancenas, 2018-10-17:
update mac80211 script 删除set wireless.radio${devidx}.noscan=1，编译正常了

hcym, 2018-10-17:
AR71
哦，可能不一样，没编译这个型号

closed 2018-11-27 by coolsnowwolf