coolsnowwolf/lede #717: samba 无法进行编译

eckholm, 2018-10-21:
编译报错如下：

touch /media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25:samba-3.6.25 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DMAX_DEBUG_LEVEL=-1 -D__location__=\\\"\\\" -ffunction-sections -fdata-sections  -I/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/usr/include -I/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/include -I/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/include -I/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include/fortify -I/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25:samba-3.6.25 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DMAX_DEBUG_LEVEL=-1 -D__location__=\\\"\\\" -ffunction-sections -fdata-sections  -I/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/usr/include -I/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/include -I/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/include -I/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include/fortify -I/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include " LDFLAGS="-L/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib -L/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/lib -L/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/lib -L/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/lib -znow -zrelro -Wl,--gc-sections " make  -C /media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25/source3 AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25:samba-3.6.25 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DMAX_DEBUG_LEVEL=-1 -D__location__=\\\"\\\" -ffunction-sections -fdata-sections" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" DYNEXP= PICFLAG= MODULES= ;
make[4]: Entering directory `/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25/source3'
Using CFLAGS     = -I../lib/zlib -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25:samba-3.6.25 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DMAX_DEBUG_LEVEL=-1 -D__location__="" -ffunction-sections -fdata-sections -I. -I/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25/source3 -I/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25/source3/../lib/popt -I/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25/source3/../lib/iniparser/src -Iinclude -I./include  -I. -I. -I./../lib/replace -I./librpc -I./.. -I./../lib/talloc -I./../lib/tevent -I../lib/tdb/include -DHAVE_CONFIG_H  -I/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/usr/include -I/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/include -I/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/include -I/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include/fortify -I/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include -Iinclude -I./include -I. -I. -I./../lib/replace -I./librpc -I./.. -I./../lib/popt  -I/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25/source3/lib -I.. -D_SAMBA_BUILD_=3 -D_SAMBA_BUILD_=3
      PICFLAG    = 
      LIBS       = -lresolv -lresolv
      LDFLAGS    = -L/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib -L/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/lib -L/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/lib -L/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/lib -znow -zrelro -Wl,--gc-sections -L./bin
      DYNEXP     = 
      LDSHFLAGS  =  -shared -Wl,-Bsymbolic  -L/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib -L/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/lib -L/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/lib -L/media/ubunt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/lib -znow -zrelro -Wl,--gc-sections -L./bin -lc -Wl,-z,defs
      SHLIBEXT   = so
      SONAMEFLAG = -Wl,-soname=
Linking bin/samba_multicall
rpc_client/cli_pipe.o: In function `rpc_api_pipe_req_send':
cli_pipe.c:(.text.rpc_api_pipe_req_send+0x287): undefined reference to `ndr_push_dcerpc_sec_verification_trailer'
../librpc/ndr/ndr_dcerpc.o: In function `ndr_pop_dcerpc_sec_verification_trailer':
ndr_dcerpc.c:(.text.ndr_pop_dcerpc_sec_verification_trailer+0x10a): undefined reference to `ndr_pull_dcerpc_sec_verification_trailer'
collect2: error: ld returned 1 exit status
make[4]: *** [bin/samba_multicall] Error 1
make[4]: Leaving directory `/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25/source3'
make[3]: *** [/media/ubunt/openwrt/lede/build_dir/target-x86_64_musl/samba-3.6.25/.built] Error 2
make[3]: Leaving directory `/media/ubunt/openwrt/lede/package/network/services/samba36'
time: package/network/services/samba36/compile#0.60#0.15#1.54
make[2]: *** [package/network/services/samba36/compile] Error 2
make[2]: Leaving directory `/media/ubunt/openwrt/lede'
make[1]: *** [/media/ubunt/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/media/ubunt/openwrt/lede'
make: *** [world] 错误 2

closed 2018-10-22 by coolsnowwolf

qq960826, 2020-04-09:
请问是啥情况？

qq960826, 2020-04-09:
我也遇到这个问题了

Jasonlixiaojian, 2020-10-12:
我也遇到了这个问题，为什么呢？急需解决